qmk/qmk_firmware #9633: wiring up a dactyl manuform

shiftux, 2020-07-03:
Hello

Quick question about wiring up a dactyl manuform split keyboard.

I was looking at the keymap of a 5x6 such as here: https://github.com/qmk/qmk_firmware/blob/master/keyboards/handwired/dactyl_manuform/5x6/keymaps/default/keymap.c
I am confused about the fact that there are 32 keys per hand in a 5x6 = 30 keys setup and i was unsure how that could be wired...

looking a bit further i found this: https://github.com/qmk/qmk_firmware/blob/master/keyboards/handwired/dactyl_manuform/5x6/5x6.h
where there actually is a 6x6 matrix defined (lower part), like so (for the left hand):

```
    { L00,   L01,   L02, L03, L04, L05 }, \
    { L10,   L11,   L12, L13, L14, L15 }, \
    { L20,   L21,   L22, L23, L24, L25 }, \
    { L30,   L31,   L32, L33, L34, L35 }, \
    { KC_NO, KC_NO, L42, L43, L44, L45 }, \
    { KC_NO, KC_NO, L52, L53, L54, L55 }, \
```

So am I correct in assuming that the wiring should follow the scheme of the above matrix? - Meaning I would wire for 6 rows and 6 cols and have col 1&2 empty on row 5 and 6? 

Additionally, the readme specifies: "Next, wire your key matrix to any of the remaining 17 IO pins of the pro micro and modify the `matrix.c` accordingly."
However I don't find `matrix.c` anywhere in the dactyl_mauform folders nor do I see a reference in the code... is there a general one or which one should I use?
There is one in `./keyboards/handwired/dactyl/matrix.c` but again, I don't see it referenced in the dactyl_manuform code...

Thanks for the help!

jptime, 2020-07-07:
That readme may have become a bit outdated.

The matrix that's used is under [quantum/split_common](https://github.com/qmk/qmk_firmware/blob/master/quantum/split_common/matrix.c)

I believe you're correct on the wiring

shiftux, 2020-07-07:
thank you @jptime 
when compiling a keyboard, there is actually command line output showing which files are being compiled and included
```
qmk compile -kb handwired/dactyl_manuform/5x6 -km my_layout
```

and the wiring worked out as intended.

best

closed 2020-07-07 by shiftux