qmk/qmk_firmware #9650: [Bug] Percent Studios Canoe Gen1 VIA hex does not handle media control

Boy-314, 2020-07-05:
## Describe the Bug
I recently saw that there was a VIA default hex for the percent studios canoe gen1. I thought I'd give it a shot, so I flashed it onto my board with the help of Bootmapper Client (couldn't figure out how or why it didn't work from my WSL terminal window)

I updated my layout through VIA so that the FN layer has media controls where the arrow keys are, but they don't seem to work. Other things on the second layer such as Delete on Backspace, arrow keys on esdf work fine though.

## System Information

 - Keyboard:  Percent Studios Canoe Gen1
   - Revision (if applicable):  
 - Operating system: Windows
 - AVR GCC version: 5.4.0

 - ARM GCC version: 8.3.1 20190703

 - QMK Firmware version: 0.9.20

 - Any keyboard related software installed? 
   - [x] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
AHK is installed on my device, but not active. No autorun scripts are installed either.

fauxpark, 2020-07-05:
V-USB boards currently cannot support VIA and media keys at the same time. It's being worked on here: https://github.com/qmk/qmk_firmware/pull/9020

WSL also does not have access to USB devices yet, so you have to use the QMK Toolbox to flash.

Boy-314, 2020-07-05:
Got it. Thanks for the link to the relevant PR.

closed 2020-07-05 by Boy-314