qmk/qmk_firmware #9665: [Feature Request] Timers between two halves of a split keyboard are not synced

Gansgar, 2020-07-06:
The main problem is, that for RGB Matrix and other uses the internal clock between both halves should be synced, which they aren't right now. As most of us don't use an RTC on their keyboard, both timers run at different speeds and can get heavily out of sync in a very short amount of time. To my understanding RGB Light circumvents this problem, by splitting the animation into multiple parts and informing the minion when a new part of the animation was started. But this concept is not feasible when using RGB Matrix. As the only other alternative that comes to my mind (sending all rgb data over the serial connection) isn't feasible easy, due to low data speeds, I'd suggest that syncing both timers is essential to the idea. 

In #9613 and #5998 a new timer class called `sync_timer` was introduced to separate the animation and internal timers. But I would suggest a general change to a synced `timer.h` to stream line the usage.  

## Description

- the master will have no alteration to their timer class
- the minion's timer class also runs normally without support, but the current time can be corrected using a function call
- the slave can request the current time from the master using (Christian's Algorithm)[https://en.m.wikipedia.org/wiki/Cristian%27s_algorithm]
  - this should occur maybe every >10ms, but not to often to prevent overload of the serial communication (i.e. every time serial loop is run)
  - on a successful timer update, minion's timer is updated accordingly using the exposed function

This description is open for discussion and is a sub issue of #9619.

## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

XScorpion2, 2020-07-20:
> But I would suggest a general change to a synced timer.h to stream line the usage.
So this is what I first did in #5998 and ran into quite a few problems. Pretty much nearly everything in QMK uses timer.h to handle quite a bit and just about any jump forward or backward in that timer, even 4ms was enough to lock the slave half and require rebooting the keyboard. From experience, I would recommend going with just the sync_timer approach and allowing systems to opt in to using it once they have been updated and tested with the sync timer and only if they really need to stay in sync. In addition, it's also worth noting that, while the timers don't stay in sync, it's doesn't matter for most systems as they are just trying to generically time start and stop events, and are not used like a real time tracker such as their use in RGB Light and RGB Matrix.

Gansgar, 2020-07-20:
Thanks. That's an interesting point. I kinda expected, but thought that the timers wouldn't run off by that much normally. But it seems they do.

closed 2020-11-19 by stale[bot]