qmk/qmk_firmware #9696: SPLIT_HAND_MATRIX_GRID helps on multiple places.

Funkest, 2020-07-10:
I think `SPLIT_HAND_MATRIX_GRID` is the best choice to use on both left and right sides usb plugging than `SPLIT_HAND_PIN`.
Because `SPLIT_HAND_MATRIX_GRID` needs no additional circuit design different than `SPLIT_HAND_PIN`, then it works on most of existing split keyboards.
This works with both used and unused intersections only for both sides using.

e.g.
Suppose intersection (D7, B2) means left side 'T' and right side 'Y'.
Add following line in some config.h.
`#define SPLIT_HAND_MATRIX_GRID D7,B2`

On plugging in left side, T must be pressed.
On plugging in right side, Y must be up.

note:
This determins the side on keyboard initialization. So the it occurs on every reset.
If it annoys on pc reboot, `#define SPLIT_HAND_MATRIX_GRID_LOW_IS_RIGHT` will be helpful.

https://github.com/qmk/qmk_firmware/blob/master/quantum/split_common/split_util.c
https://github.com/qmk/qmk_firmware/pull/8685

nooges, 2020-07-18:
Do you have a specific question or request regarding this?

Funkest, 2020-07-18:
No, I'm fully satisfied with this. Should I close this issue?

nooges, 2020-07-19:
Yep!

closed 2020-07-19 by nooges