qmk/qmk_firmware #9697: dactyl split keyboard - slave not responding

shiftux, 2020-07-10:
would love some help in debugging why the slave of my dactyl is not responding...

keyboard: dactyl-manuform 5x6
MCU: 2x pro micro
USB connected to left side, TRS jack linking right (slave) side

```
#define USE_SERIAL
#define MASTER_LEFT
```
adapted keymap and `5x6.h` and added a rotary encoder, otherwise all default code.

what i did so far:
- connected USB to the right side: this makes the right side work (with left side keys QWERT) and left side becomes unresponsive
    -> that for me means that wiring is mostly OK
- checked TRS wiring, that is fine, since power is good on slave and TRS signal is connected on pin 3 (default)

-> it seems to me that it's a software issue, but I can't pinpoint it

One thing i found weird is the code [here](https://github.com/qmk/qmk_firmware/blob/master/keyboards/handwired/dactyl_manuform/5x6/config.h) which specifies 12 rows and 6 cols, when in my understanding it should be the other way around. changing this however breaks the compilation, so i ended up leaving it...

any help is much appreciated

shiftux, 2020-07-13:
to be really sure i checked out the qmk repo clean, compiled and flashed the dactyl manuform 5x6 default keymap.
same behaviour, left side (master) is working, slave is not. if i plug the right side instead of left, it in turn behaves as if it were a left side (QWERT keys) and the other side is dead.
and i don't think it's a wiring issue. i double checked, the TRS connection pin is in PD0 (default) on both and checking the connection with a multimeter gives me the connection beep.

shiftux, 2020-07-14:
SOLVED IT !!!!!

The thread that guided me to the solution was this: https://www.reddit.com/r/olkb/comments/c0006h/dactylmanuform_wlets_split_firmware_slave_side/erb2y5l/

I ended up defining the following in `config.h`
```
#define USE_SERIAL

#define MASTER_LEFT

#define SPLIT_USB_DETECT
#define SPLIT_USB_TIMEOUT 2000

#define EE_HANDS
```

And also flashing the EEPROM to hardcode which half is left and which is right. See here for how to flash left and right sides https://leico.github.io/TechnicalNote/QMK/write-vitamins-avrdude

hope this helps someone

closed 2020-07-14 by shiftux

FoamyGuy, 2020-09-28:
Thank you for posting the soultion!

I was already defining MASTER_RIGHT, but still having trouble. These two commands you provided worked for me:
```
#define SPLIT_USB_DETECT
#define SPLIT_USB_TIMEOUT 2000
```
After adding those each half now distinguished correctly.