qmk/qmk_firmware #9717: [Feature Request]  Support Directo Macbook Pro Plugin Support

UebelAndre, 2020-07-13:
## Feature Request Type

- [ /] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ /] New behavior

## Description

I would like to be able to use the keyboard by plugging it directly into my Macbook Pro. Is it possible through firmware to make the keyboard usable if I directly plug it in via the USB-C port? (Using a MacBook Pro 2019)

MxBlu, 2020-07-13:
That's a hardware issue, not a software one.

UebelAndre, 2020-07-13:
@MxBlu Is there a place I should post this request instead?

MxBlu, 2020-07-13:
The vendor of whatever hardware you own that you want to run QMK on, USB C to C requires a bit of extra circuitry on the PCB.

UebelAndre, 2020-07-18:
Hey, sorry for the delay. Turns out this works 😄 I was using a faulty cable (the charging cable my mac came with + an adapter). Also sorry for the bother 😄

closed 2020-07-18 by UebelAndre