qmk/qmk_firmware #9733: [Bug] Could not compile with the massdrop/ctrl keyboard: multiple definition of `dmac_desc`

yu17, 2020-07-15:
The firmware could not be compiled with the massdrop/ctrl keyboard. The compilation exits with a bunch of `multiple definition of 'dmac_desc'` errors.

## Describe the Bug

#### Steps to reproduce:

Cloning a fresh copy of QMK by:

```
qmk setup
```

Then compile the firmware with:

```
qmk compile -kb massdrop/ctrl -km default
```

And the following errors would occur:

```
Linking: .build/massdrop_ctrl_default.elf                                                           [ERRORS]
 | 
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/config_led.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: multiple definition of `dmac_desc_wb'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/config_led.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: multiple definition of `dmac_desc'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/keyboards/massdrop/ctrl/keymaps/default/keymap.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: multiple definition of `dmac_desc_wb'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/keyboards/massdrop/ctrl/keymaps/default/keymap.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: multiple definition of `dmac_desc'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/common/arm_atsam/suspend.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: multiple definition of `dmac_desc_wb'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/common/arm_atsam/suspend.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: multiple definition of `dmac_desc'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/common/magic.o:/home/($HOME)/QMK/qmk_firmware.test/quantum/keycode_config.h:44: multiple definition of `keymap_config'; .build/obj_massdrop_ctrl_default/keyboards/massdrop/ctrl/keymaps/default/keymap.o:(.bss+0x4): first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/adc.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: multiple definition of `dmac_desc_wb'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/adc.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: multiple definition of `dmac_desc'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/clks.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: multiple definition of `dmac_desc_wb'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/clks.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: multiple definition of `dmac_desc'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/i2c_master.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: multiple definition of `dmac_desc_wb'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/i2c_master.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: multiple definition of `dmac_desc'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/led_matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: multiple definition of `dmac_desc_wb'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/led_matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: multiple definition of `dmac_desc'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/main_arm_atsam.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: multiple definition of `dmac_desc_wb'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/main_arm_atsam.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: multiple definition of `dmac_desc'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/spi.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: multiple definition of `dmac_desc_wb'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/spi.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: multiple definition of `dmac_desc'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:27: first defined here
 | /usr/lib/gcc/arm-none-eabi/10.1.0/../../../../arm-none-eabi/bin/ld: .build/obj_massdrop_ctrl_default/protocol/arm_atsam/usb/usb2422.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: multiple definition of `dmac_desc_wb'; .build/obj_massdrop_ctrl_default/matrix.o:/home/($HOME)/QMK/qmk_firmware.test/tmk_core/protocol/arm_atsam/i2c_master.h:28: first defined here
[... cut ...]
I had looked into this issue, and the problem is with these two files: `tmk_core/protocol/arm_atsam/i2c_master.h` and `tmk_core/protocol/arm_atsam/i2c_master.c`.

Since the issue is about duplicate definitions in the linking stage, it could actually be fixed by removing the following two lines:

```
__attribute__((__aligned__(16))) DmacDescriptor dmac_desc;
__attribute__((__aligned__(16))) DmacDescriptor dmac_desc_wb;
```

from `tmk_core/protocol/arm_atsam/i2c_master.h` and add them to `tmk_core/protocol/arm_atsam/i2c_master.c`.

However, I'm not quite familiar with the `__attribute__` thing, and am not sure if this is a proper fix for this problem.

fauxpark, 2020-07-15:
https://github.com/qmk/qmk_firmware/pull/9485

zvecr, 2021-02-27:
Fixed by 9485

closed 2021-02-27 by zvecr

chrisdonlan, 2021-05-29:
Recompiling for a keyboard that worked in 2017, and I am running into this bug again.