qmk/qmk_firmware #9744: [Feature Request] Add a key to loop between layouts

sblondon, 2020-07-17:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Instead of having several keys to enable/disable the binded layout, it would be nice to have a key to loop between layout:
each time the user hits the key, the layout is increased and at the last one, the layout number start again from 0 (layout 0 → layout 1→ layout 2 → layout 3→layout 0, etc.).

I have a ErgoDox EZ keyboard and I do not see such feature on https://configure.ergodox-ez.com. I guess the patch should be done in QMK firmware first. I'm interested to implement it if it's doable.
What do you think about it?

FilipParyz, 2020-07-26:
You can do it on a layer basis. Each layer will have a Toggle to next layer, with last layer having a Toggle to the first layer.

sblondon, 2020-07-30:
I tried your suggestion and it works well.

With the ergodox configurator, it's doable with TO key, selecting layer+1 for each layer. Last layer is set up to layer 0.
I guess this method is usable with every keyboard using QMK firmware.

Thanks for the tip!

closed 2020-07-30 by sblondon