qmk/qmk_firmware #9802: [Bug] Tap dance does not use the tapping term from get_tapping_term

Merlin04, 2020-07-24:
## Describe the Bug

Tap dance does not use the tapping term from the get_tapping_term method; it can only use either the value passed to it (if you are using ACTION_TAP_DANCE_FN_ADVANCED_TIME) or the defined TAPPING_TERM.

## System Information

 - Keyboard: Gergoplex Heavy
   - Revision (if applicable):  
 - Operating system: Ubuntu Budgie 20.04
 - AVR GCC version: 5.4.0

 - ARM GCC version: 9.2.1

 - QMK Firmware version: 0.6.212 (I'm using Germ's fork which has support for combos and the gergoplex, the version number doesn't seem right but I think the fork was from 3 months ago)

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other: 

## Additional Context

Merlin04, 2020-07-24:
The source of this appears to be in the function `process_tap_dance.c`:
```cpp
void matrix_scan_tap_dance() {
    if (highest_td == -1) return;
    uint16_t tap_user_defined;

    for (uint8_t i = 0; i <= highest_td; i++) {
        qk_tap_dance_action_t *action = &tap_dance_actions[i];
        if (action->custom_tapping_term > 0) {
            tap_user_defined = action->custom_tapping_term;
        } else {
            tap_user_defined = TAPPING_TERM;
        }
        if (action->state.count && timer_elapsed(action->state.timer) > tap_user_defined) {
            process_tap_dance_action_on_dance_finished(action);
            reset_tap_dance(&action->state);
        }
    }
}
```

drashna, 2020-07-31:
This is fixed in #6259

closed 2020-07-31 by drashna