qmk/qmk_firmware #9821: [Feature Request / Bug] Arm SPI support for RGBW LEDs

camrbuss, 2020-07-25:
The [QMK documentation](https://docs.qmk.fm/#/ws2812_driver?id=ws2812-driver) shows support for SK6812 RGBW LEDs stating that they are supported besides the use of the white led. I don't believe that this is true as the SK6812 complies with the same timing specification of WS2812 but expects 4 "bytes" to describe the 4 leds that it has on it's chip. This means that with the current ws2812_spi.c implementation only three "bytes" are sent per chip. This causes the SK6812 to become pretty confused as it is missing a "byte" for the white LED. For example, if you tell QMK that you have 4 LEDs, only 3 LEDs will ever light up.  Per the current documentation it would make sense that this would just be padded 0's but I don't see this happening 

## Feature Request Type

- [ ] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

The enhancement would simply need to add in a pre-processor macro to determine if four "bytes" need to be sent instead of three. I have already implemented the enhancement in my fork, just didn't put in a pull request because I was not sure if I was missing something or totally off on my implementation.

https://github.com/camrbuss/qmk_firmware/commit/f96461b269b91c5593aa2589b854eac1e717d7e6

camrbuss, 2020-10-27:
Do the QMK maintainers want this feature?

tzarc, 2020-10-27:
#10184 was created and was intended to cater for all other drivers, not just SPI.
At the moment it looks like that PR has stalled -- feel free to create another PR to get RGBW support in, but it'll need to cater for the same issues presented on that PR... SPI, PWM, bitbang, both ARM and AVR.

drashna, 2020-11-05:
I can't speak for others, but RGBW leds are used on some boards.  I think it's primarily the ergodox ez that does right now.  But would prefer that both AVR and ARM support was in sync.  

Additionally, the libraries that we use for WS2812 on ARM have been updated to include RGBW support, so we can add it. And I think we should.

closed 2021-03-05 by stale[bot]