qmk/qmk_firmware #9845: [Feature Request] Documentation on the include infrastructure for this project

jpmorris, 2020-07-27:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Prompting question:
Where is the definition of 'RGB_MATRIX_SOLID_REACTIVE' in the codebase? I can't find it anywhere.

Why I ask:
I'm not expert in C (mostly a python programmer in my day job), but I'm having trouble understanding this codebase.  I am would like to increase the reactive time a key stays on with the RGB_MATRIX_SOLID_REACTIVE animation mode.  So I search the code base for that definition using github or on the command line, and i can't find it anywhere.   I'm sure there is something fundamental I'm misunderstanding about how qmk works.  But this definition must be defined SOMEWHERE, and the fact that it's so hard to find must reflect a particularity about qmk that is not common among software projects and should be reflected in the documentation (sorry if it's already there, but I did look).

drashna, 2020-07-29:
It's not explicitly defined.  It's automatically generated based on the animation code. 

This block calls it in:
https://github.com/qmk/qmk_firmware/blob/0091897cc88caa0f7241def0e5d44539162cf634/quantum/rgb_matrix.c#L327-L332

This file includes all of them:
https://github.com/qmk/qmk_firmware/blob/master/quantum/rgb_matrix_animations/rgb_matrix_effects.inc

And this line is how the name is generated:
https://github.com/qmk/qmk_firmware/blob/0091897cc88caa0f7241def0e5d44539162cf634/quantum/rgb_matrix_animations/solid_reactive_anim.h#L3

It takes the `SOLID_REACTIVE` part listed here, and replaces the `##name` part in `RGB_MATRIX_##name`. 

It's some fancy/tricky C preprocesser stuff, but it's pretty simple. 

Adding new animations is actually covered here:
https://docs.qmk.fm/#/feature_rgb_matrix?id=custom-rgb-matrix-effects

The process is identical for each, it's just a matter of which `inc` file you add the animation to.

jpmorris, 2020-07-31:
Thanks with your help I was able to simply modify the startup speed:

```#define RGB_MATRIX_STARTUP_SPD 10```

in config.h
Thanks so much!

closed 2020-11-29 by stale[bot]