qmk/qmk_firmware #9868: [Feature Request | Docs] Add MIDI keycodes to the Full List of Keycodes

precondition, 2020-07-29:
## Addition to the documentation

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

[The full list of keycodes](https://beta.docs.qmk.fm/using-qmk/simple-keycodes/keycodes) is missing all the MIDI keycodes found [here in the `quantum_keycode.h` file](https://github.com/qmk/qmk_firmware/blob/master/quantum/quantum_keycodes.h#L166).

I'd submit a PR myself but I am not familiar at all with MIDI or music in general, really, so I won't be able to write a description of all those keycodes.

wxyangf, 2021-06-05:
I'll take a look at this after the main MIDI doc issue has been merged

fauxpark, 2021-07-13:
Looks like this has been fixed by the above PR. Closing.

closed 2021-07-13 by fauxpark