qmk/qmk_firmware #9906: [Bug] QMK doctor only checks /etc/udev/rules.d for udev rules.

bbaserdem, 2020-08-01:
The doctor command in qmk python library fails to find the correct rules in `/usr/lib/udev/rules.d` directory.

## Describe the Bug

The doctor command in QMK python library only checks a hard coded `/etc/udev/rules.d` for udev rules. Which works fine if you put your rules there. But these rules can also be located in `/usr/lib/udev/rules.d`; and actually is the preferred location for packages to drop their udev rules. I recently packaged the [python client in archlinux](https://aur.archlinux.org/packages/python-qmk); and also provided neccessary udev rules along with it. After switching to these rules, I noticed I was getting warnings from `qmk doctor` when I had the correct udev rules in place.

## System Information

 - Keyboard: N/A
 - Operating system: Archlinux
 - AVR GCC version: 10.1.0
 - ARM GCC version: 10.1.0
 - QMK Firmware version: 0.9.46
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I also implemented a quick fix; and will issue a pull request.

zvecr, 2020-08-01:
As a note <https://aur.archlinux.org/packages/qmk> exists and mentioned in the [install docs](https://docs.qmk.fm/#/newbs_getting_started) `qmk package from AUR (note: it’s maintained by a community member): yay -S qmk)`.

bbaserdem, 2020-08-01:
@zverc The aforementioned package stealth installs `python-hjson` python package without seperately packaging it (very bad practice) and uses pip to do so (another bad practice). After I contacted the maintainer without response, (granted I did not wait longer than a day) I repackaged both `python-qmk` and `python-hjson` in the AUR to fit archlinux packaging guidelines better.

Foxboron, 2020-09-12:
Honestly, is hjson needed? It's used one place to somehow streamline the parsing of some JSON data. But it seems like a rather unorganized project with no releases since 2014. Could we check if it's possible to remove it?