qmk/qmk_firmware #9923: having a macro repeat until release of key

Draic, 2020-08-03:
Sorry if this is not the appropriate place for this question, but I cannot seem to find anything regarding this.

I have a macro which just sends a sequence of button presses for a tedious task I have to do.

`    case grmacro:
      if (record->event.pressed) {
        SEND_STRING(SS_TAP(X_BTN1));
        _delay_ms(200);
        SEND_STRING(SS_TAP(X_ENTER));
        _delay_ms(350);
        SEND_STRING(SS_TAP(X_ENTER));
        _delay_ms(350);
        SEND_STRING(SS_TAP(X_DOWN));
        _delay_ms(350);
        SEND_STRING(SS_TAP(X_ENTER));
        _delay_ms(350);
        SEND_STRING(SS_TAP(X_ENTER));
        _delay_ms(350);
        SEND_STRING(SS_TAP(X_ENTER));
        _delay_ms(350);
        SEND_STRING(SS_TAP(X_ENTER));
        _delay_ms(350);
        SEND_STRING("10");
        SEND_STRING(SS_TAP(X_ENTER));
        _delay_ms(350);
        SEND_STRING(SS_TAP(X_ENTER));
        _delay_ms(350);
        SEND_STRING("5");
        SEND_STRING(SS_TAP(X_ENTER));
        _delay_ms(350);
        SEND_STRING(SS_TAP(X_ENTER));
        _delay_ms(350);
        SEND_STRING(SS_TAP(X_ENTER));
        _delay_ms(350);
        SEND_STRING(SS_TAP(X_ENTER));
      }
      break;`

My issue is that I want this to repeat until I release the key that triggered the macro. I press the key and this macro should get looped over and over until I release the key. I can not find anything similiar to this. Is this possible to get working in QMK?

closed 2021-01-07 by stale[bot]

yeldiRium, 2021-02-08:
I have the same problem outlined in this issue. The stale bot closed it, since there were no replies here, but the problem still stands and there is nothing to add to the question. Can this please be re-opened?

Draic, 2021-02-08:
I also still hope someone can help with his :(

spidey3, 2021-02-08:
It's doable. In `process_record_user()`, set a boolean variable when the key is pressed, and clear it when it is released.  In matrix_scan_user(), do the SEND_STRING if the boolean is true. That way on every cycle through the matrix scan the desired strings will be sent. You might want to check against a timer so that you send only every few ms, instead of on every scan.  

That being said, I'm not sure that QMK is the best place to implement this. because you are basically re-implementing auto-repeat, which is actually a host-side functionality. It might be better done in a host-side tool like Karabiner or AutoHotkey – in which case you would use the host-side tool to map some key combination (like Alt-M) to the string you want...

yeldiRium, 2021-02-09:
@spidey3 thank you for the response!

I have a follow-up question, if you're so kind: If auto-repeat is a host-side functionality, then the host must in some way recognize repeatable inputs. In my case I am using `SEND_STRING(SS_TAP(X_DOT));` to send a `.`. Shouldn't the host realize that this is a repeatable input? And if not, do you know a way to make this repeatable by the host?

P.S.: I'm sending the `.` in this complicated manner because I need to customize the sent characters when `shift` is held.

spidey3, 2021-02-09:
The host determines whether to auto-repeat based on whether it sees a KeyDown event without a matching KeyUp event.  If there is a long enough period without seeing a matching KeyUp, the host interprets it as an auto-repeat, and then stops repeating when it sees the KeyUp.

`SEND_STRING(SS_TAP(X_DOT));` is sending a tap (KeyDown(X_DOT) then KeyUp(X_DOT) - so the keyboard doesn't recognize it as something to repeat. You need to change your code to sent the KeyDown(X_DOT) when `record->event.pressed == true`, and send the KeyUp(X_DOT) when `record->event.pressed == false`.

Now that I've said that, I have a sneaky suspicion that you this send string is all about sending a unicode character.  There's a [much better way to do that](https://docs.qmk.fm/#/unicode) that doesn't require code in `process_record_user()` for every unicode character that you want to send, and that will work across host types.

nathanvercaemert, 2021-12-19:
This was super helpful. Just to clarify a use case for this kind of behavior: I'm using it in conjunction with AHK to achieve a smooth scrolling experience when I need to move the whole window instead of just scrolling the contents. Having all of the events firing as rapidly as possible from all places gave me the freedom to tweak the behavior for the best experience.