qmk/qmk_firmware #9928: Using clang instead of gcc

heshananupama, 2020-08-03:
I am currently using arm-none-eabi-gcc toolchain to compile the QmK firmware (firmware related to arm architecture instead of avr) my question is whether I could use llvm toolchain instead of GCC. If possible, how could we do it?

drashna, 2020-08-12:
there are some settings for the makefile that you can specify a different compiler.  However, the biggest issue/hurdle is that a lot of the code may not compile outright on clang.

nektro, 2022-02-24:
old does in fact not mean resolved

tzarc, 2022-02-25:
> old doesn't mean resolved

We have to pick our battles -- adding support for clang is best left to someone in the community. Perhaps that's you?

nektro, 2022-02-25:
perhaps, I was trying to get stale bot to reopen the issue so folks know the task is available

tzarc, 2022-02-25:
Guess we'll see if it goes stale again, then.

tzarc, 2022-06-12:
Short-circuiting stalebot. Closing. If someone's keen to get clang working, they'll do so.

closed 2022-06-12 by tzarc