qmk/qmk_firmware #9959: New handwired keyboard - split ortho.

FilipParyz, 2020-08-07:
I'd like to add my handwired keyboard to the mix of keyboards in the qmk.
Finished keyboard (soldered, programmed, working) can be seen on this imgur post: https://imgur.com/a/9UyvhUZ
It comes with a simple default keymap, it's a split ortho, two simetrical sides with 5x6 grid with 2u keycaps under thumbs. Very simple layout. I was originally inspired by Preonic ortho.
I've made my own version (handwired) of Preonic years ago, but didn't push it to the QMK. Now I want to fix my mistake by uploading a new keyboard!

I forked qmk and I'm working on that keyboard on branch [paryzSplit56](https://github.com/FilipParyz/qmk_firmware/tree/paryzSplit56). It's working well, I may just add RGB underglow to it.
When I'm finished, should I merge that branch to my own master, then to qmk, or from my branch to qmk?

kb-elmo, 2020-08-07:
you can just create a pullrequest directly from your branch to the qmk master.

you won't be able to merge it yourself though. you have to go through the standard review process for that.

FilipParyz, 2020-08-07:
Oh no, I meant a PullRequest, I just wasn't sure if I need to merge to master on my side. Thanks for clarification :)

FilipParyz, 2020-11-06:
Oh, @kb-elmo I've added my keyboard to the QMK :D
[This PR](https://github.com/qmk/qmk_firmware/pull/10724) was mine. Thanks for your help :)

closed 2020-11-06 by FilipParyz