qmk/qmk_firmware #9969: Cant create a new keyboard default keymap

Lil-BenQ, 2020-08-08:
On step 4 of  https://beta.docs.qmk.fm/tutorial/newbs_getting_started#4-test-your-build-environment ,its tells me to test my build environment by inputting the code : qmk compile -kb <keyboard> -km default. When I do, I get this response from msys64;

$ qmk compile -kb test75/rev1 -km default
INFO Compiling keymap with make test75/rev1:default

QMK Firmware 0.9.49
make: *** No rule to make target 'test75/rev1:default'. Stop.
|
|  QMK's make format recently changed to use folder locations and colons:
|     make project_folder:keymap[:target]
|  Examples:
|     make dz60:default
|     make planck/rev6:default:flash

which is different from what QMK's guide is saying i should get.

fauxpark, 2020-08-08:
`<keyboard>` must be an already existing keyboard in the repo. It does not create a new keyboard & keymap config, it just compiles one.

Lil-BenQ, 2020-08-12:
Alrighty I see! How would I create a new keyboard and keymap configuration then? Sorry I'm really new to this!

Lil-BenQ, 2020-08-24:
@nblyumberg yes i am creating a new pcb! what would be the best way for me to go about making the firmware? I was doing some reading online and people were saying to just copy other keyboards firmware that are similar to mine and make adjustments. i havent been able to find one that uses a QMK Proton C board yet though. :(

nblyumberg, 2020-08-24:
Take a look at Clueboard 66 rev 4. That uses the STM32F303

closed 2020-12-23 by stale[bot]