qmk/qmk_firmware #9974: Questions about Leader key feature

NMHoang05, 2020-08-08:
Hello! 
I want to ask about: https://docs.qmk.fm/#/feature_leader_key

Q1) 
to use Leader key feature, I need to define a `LEADER_DICTIONARY`, right?
is there a generic implementation that would work without defining the dictionary?

I mean, for example:
- (press these in sequence)  `Leader, Ctrl, C`
     (then) leader key timed out:  send `Ctrl + C`
- (press these in sequence)  `Leader, Ctrl, Shift, V`
     (then) leader key timed out:  send `Ctrl + Shift + V`

(above sequences are not defined in dictionary)

Q2) 
it seems while waiting for sequence, `MO()` is ignored.

I have key `5` on normal layer and `F5` on Fn layer
below is the debug output from QMK Toolbox

I hold `MO(1)`, press `5` then release them.
```
  > KL: kc: 20739, col: 6, row: 3, pressed: 1
  > KL: kc: 62, col: 5, row: 0, pressed: 1
  > KL: kc: 62, col: 5, row: 0, pressed: 0
  > KL: kc: 20739, col: 6, row: 3, pressed: 0
```

I press `KC_LEAD`, release it then hold `MO(1)`, press `5` then release them.
```
  > KL: kc: 23575, col: 7, row: 3, pressed: 1
    leader start
  > KL: kc: 23575, col: 7, row: 3, pressed: 0
  > KL: kc: 20739, col: 6, row: 3, pressed: 1
  > KL: kc: 34, col: 5, row: 0, pressed: 1
  > KL: kc: 34, col: 5, row: 0, pressed: 0
  > KL: kc: 20739, col: 6, row: 3, pressed: 0
  > leader end 20739  34  0  0  0
```
numbers in the last line (leader end...) are values of array `leader_sequence`
as you can see, the keycode stored in `leader_sequence` is 34 (`5`) instead of 62 (`F5`)

is there a way for the leader sequence to recognize keycode 62 instead of 34?

Yatreyu, 2020-08-14:
For Q2: It sounds like you want to add [Strict-Key-Processing](https://beta.docs.qmk.fm/using-qmk/advanced-keycodes/feature_leader_key#strict-key-processing) so the MO(1) won't disqualify your sequence.

For Q1: I haven't seen this. I have put these into my dictionary as: 
`//Paste function`
      `SEQ_ONE_KEY(KC_S)`
      `{`
      `register_code(KC_LCTL);`
      `register_code(KC_V);`
      `unregister_code(KC_V);`
      `unregister_code(KC_LCTL);`
      `}`

NMHoang05, 2020-08-18:
Hi, it seems Strict-Key-Processing isn't what I need.

I tried to make a generic implementation

here is what I have
```
void leader_start(void) {
  print("leader start\n");
}

void leader_end(void) {
  if (leader_sequence[0] == 0 && leader_sequence[1] == 0 && leader_sequence[2] == 0 && leader_sequence[3] == 0 && leader_sequence[4] == 0) return;

  for(int i = 0; i < 5; ++i) {
      if (leader_sequence[i] != 0) register_code(leader_sequence[i]);
  }
  for(int i = 4; i >= 0; --i) {
      if (leader_sequence[i] != 0) unregister_code(leader_sequence[i]);
  }
  uprintf("leader end %u  %u  %u  %u  %u   \n", leader_sequence[0], leader_sequence[1], leader_sequence[2], leader_sequence[3], leader_sequence[4]);
}
```
it works but I realize I only need basic keycode and modifier so I change the `process_leader` in core a bit (as I don't know how to override this function)
```
bool process_leader(uint16_t keycode, keyrecord_t *record) {
    // Leader key set-up
    if (record->event.pressed) {
        if (leading) {
            if (timer_elapsed(leader_time) < LEADER_TIMEOUT) {
#    ifndef LEADER_KEY_STRICT_KEY_PROCESSING
                if ((keycode >= QK_MOD_TAP && keycode <= QK_MOD_TAP_MAX) || (keycode >= QK_LAYER_TAP && keycode <= QK_LAYER_TAP_MAX)) {
                    keycode = keycode & 0xFF;
                }
#    endif  // LEADER_KEY_STRICT_KEY_PROCESSING
#    ifdef LEADER_KEY_BASIC_AND_MOD_ONLY
                if (keycode > QK_BASIC && keycode < QK_MODS_MAX) {
#    endif // LEADER_KEY_BASIC_AND_MOD_ONLY
                if (leader_sequence_size < (sizeof(leader_sequence) / sizeof(leader_sequence[0]))) {
                    leader_sequence[leader_sequence_size] = keycode;
                    leader_sequence_size++;
                } else {
                    leading = false;
                    leader_end();
                }
#    ifdef LEADER_KEY_BASIC_AND_MOD_ONLY
                }
#    endif // LEADER_KEY_BASIC_AND_MOD_ONLY
#    ifdef LEADER_PER_KEY_TIMING
                leader_time = timer_read();
#    endif
                return false;
            }
        } else {
            if (keycode == KC_LEAD) {
                qk_leader_start();
            }
        }
    }
    return true;
}
```
I added `LEADER_KEY_BASIC_AND_MOD_ONLY`.

so far it works for the default layer but as I have mentioned, I tried to get `F5` in leader key sequence by pressing `MO(1) + 5` but that did not work.

is there a way to reference to `PROGMEM` in `keymap.c`? 
if it's not possible to do get such reference (because `process_leader` is in the core), is there a way I can override `process_leader` in my keyboard's code? (like how we override `leader_start` and `leader_end`)

closed 2020-12-16 by stale[bot]