qmk/qmk_firmware #9985: [Bug] WS2812 driver with Tim3 Audio enabled causes hard lock on AVR

drashna, 2020-08-09:
## Describe the Bug

If you enable audio on any of the Tim3 pins (B5-7), and enable WS2812 (either rgblight or rgb matrix using ws2812) it causes the controller to hard lock on initialization.  

Tested on a pro micro and elite C, noticed previously on an atmega32u2. 

## System Information

 - Operating system: MacOS 10.15.6
 - AVR GCC version: 8.4.0
 - ARM GCC version: N/A
 - QMK Firmware version: 0.9.50

sigprof, 2020-08-19:
What are the exact steps to reproduce the problem? My attempt to reproduce this with a 5V Pro Micro and the slightly modified `handwired/onekey/promicro` code (added `#define B5_AUDIO` and even `#define AUDIO_CLICKY`, compiled with `make handwired/onekey/promicro:rgb:flash LTO_ENABLE=yes AUDIO_ENABLE=yes SLEEP_LED_ENABLE=no COMMAND_ENABLE=no`) did not show any problems.

drashna, 2020-10-06:
I used an Elite C and Pro Micro. 

```make
RGBLIGHT_ENABLE = yes
AUDIO_ENABLE = yes
```

RGB pin set to F6. 

Set the audio to b5, b6 or b7, 

I tested this with a Launchpad macropad, initially.

closed 2024-10-13 by drashna