qmk/qmk_firmware #9996: Newbie question: what exactly is a layer

iandoug, 2020-08-10:
Hi

What exactly is a layer?

Or put another way: are the lower case letters typically the base layer, and their shifted versions (upper case) typically layer 1?

I'm asking because I'm trying to figure out how to divorce the typical ANSI pairings. At the moment for the NumRow, but I may want to put two lower case letters on same key (and their capse elsewhere). 

As I read the docs, it seems that defining a key to be for example "A" implies both lower and upper case at the same time.

Or am I missing something?

Thanks, Ian

quisys, 2020-08-10:
Ask on r/olkb, this isn't a good place.

iandoug, 2020-08-10:
I came here by following a link in the QMK docs where they wanted to know how to improve the docs.

Didel, 2020-08-13:
There are a few important concepts to understand:

> As I read the docs, it seems that defining a key to be for example "A" implies both lower and upper case at the same time.

A keyboard doesn’t exactly send out any string of text, instead it sends out a ‘keycode’. In stead of sending ‘a’ as a value, you should compare it more to something like: “The user pressed the key identified as `0x41`“. In the ANSI layout that is the second key of the third row (or 4th row if you’d count the F-keys). If you’ve ever messed around in the keyboard settings of your computer, you might have seen the option to choose a different keyboard ‘layout’, for example the AZERTY layout (used in France). Now, imagine having a physical AZERTY keyboard, but you didn’t change the software to match and it is still configured to assume a QWERTY keyboard. If you’d press the physical ‘a’ key on your AZERTY keyboard, you’d then get a ‘q’ on the screen because thats what the software thinks you are trying to type.
(Side note: if I’m not mistaken, you can actually have QMK send strings in stead of keycodes, so you could have a whole word or sentence typed when you press a single key, but that’s not the default way it works).

Then, what about the capital ’A’? Well, if you type a capital A, you actually press both the Shift and the A keys. And that’s exactly what is sent to the computer: “User presses these two keys simultaneously”. The software on your computer basically checks if you press shift (or any other modifier key) and a letter simultaneously, and if so, it fills in a capital A instead of the regular letter for you. So, there is no keycode for ‘capital a’.

And last but not least: Layers. Now this is something more specific to QMK, and is not applicable to all keyboards (AFAIK). Obviously, you can’t have an endless amount of physical keys on your keyboard, as space and the number of keys you can comfortably type on is limited. You might have seen that there are ‘[60%](https://blog.wooting.nl/wp-content/uploads/2018/01/60-percent-keyboard-form-factor-min.png)’ or even ‘[40%](https://www.keebtalk.com/uploads/db8059/original/1X/32e6dc25153a1cc2f4af3fbfca3d21f69aeb7173.png)’ keyboards, that actually miss a lot of physical keys you’d have on a ‘regular’ keyboard. But that doesn’t necessarily mean they miss the functionality. Think of a layer as stacking another keyboard on top of the existing one. You ‘sacrifice’ one key on the keyboard to activate the new layer, but then in return you get a whole layer worth of new keys you can assign whatever you want to. For example, you might have a second layer where the arrow keys are mapped to media controls (up/down for volume + and -, left and right for previous and next song). Or you could have a separate layer to place all the different RGB lightning controls on. Or you could have a separate layer that maps an entirely different keyboard layout, for example the Dvorak layout. The main benefit of the layer is that (unless configured otherwise), you’ll need to press the activation key to activate the layer, so you won’t accidentally press the other layers’ keys. If I remember correctly you could stack up to 15 layers in QMK, so even with just a few physical keys you can have a full keyboard experience. Also, because the layer is stored on the keyboard rather than in software on your computer, you can always use them on your keyboard, even if it is connected to a different computer. One more thing you need to know: layers can have ‘transparent’ keys on them. That means that you haven’t defined a specific keycode for that key in that layer. If you’d still press that key, it will fall back to the first layer below it that does have a keycode configured. That’s why your first (or base) layer should define a keycode for every single key.

> I may want to put two lower case letters on same key (and their capse elsewhere)

So, to your actual question. If I understand correctly, you want to have one key that when you press shift, instead of capitalizing the same letter types a different letter. With the knowledge above, that should be relatively easy: have a ‘normal’ base-layer, and configure your shift key to act as a layer change key. Then, on the second layer, add the keycode for the different letter you want and optionally the other ‘capital’ letters as well. However, I don’t know exactly if that would work out, because the shift key won’t function as a shift key any more. But perhaps you can map that to the right shift key, or a different key that would be more suitable.

There are lots of other things to tell, and maybe there are some mistakes in my story, but I hope you get the point and have a better idea of how layers work :)

iandoug, 2020-08-14:
Hi Didel

Thanks for your long and detailed reply.
I'm not a total noob at keyboards (I own keyboard-design.com) , just with QMK. I should have been more explicit in my initial query.

The part that I'm struggling to understand is (mmm how to put this....) ... the difference between implicit and explicit layers.

For example, when I test layouts on Keyboard Layout Analyzer, I specifically say what character is unshifted, what character is shifted, and even what's on AltGr and Shift-AltGr.

But when I read various QMK keymaps, all I see is upper case letters... there is no differentiation between upper case and lower case. Now I understand that for normal ANSI style layouts, somewhere between QMK and the OS (in my case, Linux), there is some agreement about what happens when you press Shift and the letter a (for example, on a QWERTY layout) so that you get "A".

This same "implicit" agreement seems to be in place for what happens when you press shift and comma.

But many alternate keyboards don't have left angle on the comma, they have something else.

So I would expect keymaps to explicitly have a base layer, and a shift layer, and an AltGr layer, and a shifted AltGr layer, and then whatever other Hyper/SuperDuper/Magic/etc layers people want to stack on top of those.
At the moment is seems like the Shift layer is almost completely IMPLICIT in the keymaps. 

Or am I overcomplicating this? I just think that stating explicitly what happens with every key on every layer would be clearer...

So I should just add a shift layer, and most of the entries (on for example, normal English layout) will be "transparent", except for where I explicitly want a different character?

But what happens when my base layer symbol is "?" and my shifted symbol is "/" ? Does QMK send "user pressed "/?" plus shift, and then it gives me ? instead of / ?

Hence why I think making everything explicit would be clearer ...

Thanks, Ian

drashna, 2020-08-14:
Ah, then the difference here is the layers in QMK, vs the layout on the host system. 

Each layout/language has different ways it handles this, with ... creates issues for QMK.  Specifically, QMK Firmware acts as a HID keyboard device, which means it sends HID Scan codes to the system, which then interprets those into keycodes.  And by default, it assumes English (US).

Various community members, and especially fauxpark have put together mappings for various keyboard layouts:
https://docs.qmk.fm/#/reference_keymap_extras

If you include these and use the keycodes within them, you can get a layout that is closer to what you're looking for.

alaricljs, 2020-09-01:
Shift / Ctrl / Alt  (and more... ) are Modifiers.  When they are pressed and held in combination with other keys the OS on the computer the KB is attached to will modify its behavior according to its keymap.

Layer keys in QMK (and others) modify the behavior of the keyboard.  Hold down MO(1) and you are telling the keyboard to switch its interpretation of the switches being pressed to a different table of information within the kb firmware.  That means that you can have the switch with the 'A' keycap send the keycode for Insert when MO(1) is held.

These two completely different behaviors can seem very similar from the perspective of the computer, however they are quite different due to what device is responsible for the behavior.  I don't know of a way to tell Windows that shift-a should come out as the 1 character.  There is in fact a way to do that in XOrg on Linux.  It doesn't matter what keyboard you are using if you fiddle with that behavior on the computer.  When you move to a new computer you don't retain that behavior without additional setup.

On your comment about a base layer symbol of '?' and shifted of '/':  While probably achievable with QMK (never tried it due to a lack of need), the solution will send "lies" to the computer.  Pressing ? on your base layer will have to send two keycodes, "shift" and "/" to the computer and pressing shift-? will have to send "/" to the computer and the computer will have to be configured to interpret those key presses 'normally' for the result to be correct.

There are some extra keycodes that may be making you think the above is not entirely true.  For instance there are 2 keycodes for the '\  |' combo, one is for ISO keyboards located between left shift and Z, the other for ANSI boards located approximately between backspace and Enter.  Use the QMK web configurator and toggle between the ANSI and ISO keycode tabs and you can compare.  Interestingly, ", <" is not one of these but that ISO '\ |' is sometimes mapped to '< >' in the OS keymap.  
As such, depending on the keymap set on the computer sending those 2 '\ |' keycodes (0x31 and 0x64) back to back won't necessarily end up in 2 '\'s showing up.  This is no different than the behavior difference between '5 %' and the keypad '5' on the same keyboard.  Different keycodes that get interpreted by the computer differently: shift-keypad_5 isn't %

AltGr is a different sort of special, it's actually just the Right Alt keycode unless it's being emulated in which case its Ctrl and Alt together.  If you really wanted to you could get different behavior from the right vs left on all modifiers since they are indeed different keycodes.  Screw with your OS keymap customizing friends by gifting them a QMK keyboard with the modifiers reverse mapped!  AutoHotKey on Windows can also be used to recognize the different keycodes and handle them for hotkeys and other things.

So if it's not apparent, in some ways you're under-complicating things.

iandoug, 2020-09-02:
Hi

Thanks for the long explanation. I'm on Linux, and realised the other night it would be simpler to let xkb do the different assignment rather than trying to make QMK do them. So the computer will think it's more or less a plain vanilla ANSI attached.

I just need QMK to turn on an led when I'm in "Insert" mode.

Thanks, Ian

closed 2020-09-02 by iandoug