qmk/qmk_firmware #10005: [Bug] Including "sendstring-dvorak.h" induces compilation error

bbaserdem, 2020-08-11:
I have been following the QMK documentation about [adapting macros to software keymap](https://docs.qmk.fm/#/feature_macros?id=alternative-keymaps). However; when I `#include "sendstring_dvorak.h"`; I'm getting a linker error; from multiple definitions of `ascii_to_keycode_lut`.

I assume there is something funky going on; or the documentation is lacking in instructions as to where this file needs to be included.

## Describe the Bug

When I include the aforementioned line in my <userspace>.h; I get the following error on the Linking line;

```
Linking: .build/planck_light_bbaserdem.elf                                                          [ERRORS]
 | 
 | /usr/bin/avr-ld: .build/obj_planck_light_bbaserdem/bb-macro.o (symbol from plugin): in function `process_record_macro':
 | (.text+0x0): multiple definition of `ascii_to_keycode_lut'; .build/obj_planck_light_bbaserdem/bbaserdem.o (symbol from plugin):(.text+0x0): first defined here
 | /usr/bin/avr-ld: .build/obj_planck_light_bbaserdem/bb-audio.o (symbol from plugin): in function `layer_state_set_audio':
 | (.text+0x0): multiple definition of `ascii_to_keycode_lut'; .build/obj_planck_light_bbaserdem/bbaserdem.o (symbol from plugin):(.text+0x0): first defined here
 | /usr/bin/avr-ld: .build/obj_planck_light_bbaserdem/bb-keylight.o (symbol from plugin): in function `keylight_set_left':
 | (.text+0x0): multiple definition of `ascii_to_keycode_lut'; .build/obj_planck_light_bbaserdem/bbaserdem.o (symbol from plugin):(.text+0x0): first defined here
 | /usr/bin/avr-ld: .build/obj_planck_light_bbaserdem/layouts/community/ortho_4x12/bbaserdem/keymap.o (symbol from plugin): in function `keylight_set_left':
 | (.text+0x0): multiple definition of `ascii_to_keycode_lut'; .build/obj_planck_light_bbaserdem/bbaserdem.o (symbol from plugin):(.text+0x0): first defined here
 | lto-wrapper: warning: Options to Xassembler do not match: -adhlns=.build/obj_planck_light_bbaserdem/bbaserdem.lst, -adhlns=.build/obj_planck_light_bbaserdem/bb-macro.lst, dropping all -Xassembler and -Wa options.
 | collect2: error: ld returned 1 exit status
 | 
``` 

## System Information

 - Keyboard: Planck
   - Revision (if applicable): light
 - Operating system: Archlinux (and tried with docker)
 - AVR GCC version: Both in 10.2.0 and in the docker build

 - ARM GCC version: 

 - QMK Firmware version: 0.9.50

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I'm getting this error both on the docker and on my operating system that uses gcc 10.2.0
My fork for my userspace code is [here](https://github.com/bbaserdem/qmk_firmware/tree/sbp-keymap/users/bbaserdem)

bbaserdem, 2020-08-13:
So apparently; I need to have the include on a .c file; as putting it in a header file defines it multiple times (even with pragma once) and makes compilation fail due to this being a const variable.

I think this should be specified in the documentation.

closed 2020-08-13 by bbaserdem

fauxpark, 2020-08-13:
`#pragma once` only ensures the header is included [once per source file](https://en.wikipedia.org/wiki/Pragma_once), it does not mean the header is included  once for the entire project.