qmk/qmk_firmware #10025: [Feature Request] Hot-switch "master" side of split keyboard

jackphilippi, 2020-08-13:
Hi all. I'm running a ReDox kb which has a pro micro driving each side. I'm wondering if anybody has some immediate ideas on how one could have a trigger to switch the master side from left to right ad-hoc. 

As this is currently managed via macros `#define`. Logistically, how hard would it be to have these set up dynamically?

My use case:

As a user, I want to be able to plug a USB cable into each side of my keyboard - one into my work laptop and one into my desktop PC - so that I can easily switch between the two devices at the press of a button. 

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

FilipParyz, 2020-10-13:
So you want a KVM switch but INSIDE the keyboard. That's a neat idea with a very niche use case.
I'll think and check what I can do about it :D

dpapavas, 2020-11-20:
> As a user, I want to be able to plug a USB cable into each side of my keyboard - one into my work laptop and one into my desktop PC - so that I can easily switch between the two devices at the press of a button.

Not sure what button you refer to, but if you intend to plug boths sides into powered USB ports at the same time, I would advise making very sure that this is "allowed" by the hardware.  Generally in split keyboards the master will power the slave over the interconnecting cable.  Unless the boards have proper switching hardware to block the +5V from reaching the slave's USB port, this will connect the power/ground lines of the two ports, which may potentially have undesirable side-effects.

If the push of said button resets both sides (i.e. first disconnects them from one computer, causing them to power down, then connects them to the other, powering them back up from the other side), then you can already achieve what you want.  Look [here](https://beta.docs.qmk.fm/using-qmk/hardware-features/feature_split_keyboard#handedness-by-eeprom).  You'll only need to flash the EEPROM of each side once and it will permanently set the left/right side, so you can connect either to the host and it will work properly.

GreyAsteroid, 2021-03-03:
I'm holding off creating a new feature request because I believe this one encompasses what I'm looking for (massdrop alt/ctrl support a host connection on both ports but there's no way to switch between them via a keypress).

closed 2021-07-08 by stale[bot]