qmk/qmk_firmware #10027: [Bug] Unicode input modes are insensitive to

bbaserdem, 2020-08-13:
When using unicode mappings with non-qwerty software keymaps with linux; the unicode entry mode does note work.

## Describe the Bug

The joint unicode modes has CTRL U hardcoded in the [process_unicode_common.h](https://github.com/qmk/qmk_firmware/blob/master/quantum/process_keycode/process_unicode_common.h).

For regular keymap qwerty; this is no problem.

If an alternative software keymap is used; the string sent for the unicode point (for an ellipsis `…` ; this would be `2026`) is correctly translated. But the `UNICODE_KEY_LNX` does not adapt to the keymap; and the unicode entry mode does not work with any keymap that maps the U key on qwerty to some other letter.

## System Information

 - Keyboard:  All
   - Revision (if applicable):  all
 - Operating system: Linux (archlinux)
 - AVR GCC version: <irrelevant>

 - ARM GCC version: <irrelevant>

 - QMK Firmware version: 0.9.50

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I can implement and submit a PR to fix this; if anyone can suggest a method to fix this; but I am not too familiar with the inner workings to do this in a proper way.

fauxpark, 2020-08-13:
It's not hardcoded, you can change it by defining `UNICODE_KEY_LNX` in your config.h. Is that not working?

bbaserdem, 2020-08-13:
The override works.

This behavior is not documented anywhere. Maybe the keymaps should redefine this character?

fauxpark, 2020-08-13:
From the [Unicode docs](https://docs.qmk.fm/#/feature_unicode?id=input-modes):
> By default, this mode uses Ctrl+Shift+U (LCTL(LSFT(KC_U))) to start Unicode input, but this can be changed by defining UNICODE_KEY_LNX with a different keycode

bbaserdem, 2020-08-13:
Oversight in my part for sure then.

closed 2020-08-13 by bbaserdem