qmk/qmk_firmware #10041: RGB Light Segment instructions do not compile.

Yatreyu, 2020-08-14:
https://beta.docs.qmk.fm/using-qmk/hardware-features/lighting/feature_rgblight

For the example code in the QMK docs for RGB Segments won't compile:

`// Light LEDs 6 to 9 and 12 to 15 red when caps lock is active. Hard to ignore!`   

    const rgblight_segment_t PROGMEM my_capslock_layer[] = RGBLIGHT_LAYER_SEGMENTS(
    {6, 4, HSV_RED},       // Light 4 LEDs, starting with LED 6
    {12, 4, HSV_RED}       // Light 4 LEDs, starting with LED 12);`
`// Light LEDs 9 & 10 in cyan when keyboard layer 1 is active  `

    const rgblight_segment_t PROGMEM my_layer1_layer[] = RGBLIGHT_LAYER_SEGMENTS(
    {9, 2, HSV_CYAN);`
`// Light LEDs 11 & 12 in purple when keyboard layer 2 is active  `

        const rgblight_segment_t PROGMEM my_layer2_layer[] = RGBLIGHT_LAYER_SEGMENTS(
        {11, 2, HSV_PURPLE});`
`// etc..` 

Then:

`// Now define the array of layers. Later layers take precedence`
        `const rgblight_segment_t* const PROGMEM my_rgb_layers[] = RGBLIGHT_LAYERS_LIST(`
       ` my_capslock_layer,`
       ` my_layer1_layer,    // Overrides caps lock layer`
        `my_layer2_layer     // Overrides other layers);`

`void keyboard_post_init_user(void) {`
    `// Enable the LED layers`
    `rgblight_layers = my_rgb_layers;}`

Then: 

`layer_state_t layer_state_set_user(layer_state_t state) {`
    `// Both layers will light up if both kb layers are active`
    `rgblight_set_layer_state(1, layer_state_cmp(state, 1));`
    `rgblight_set_layer_state(2, layer_state_cmp(state, 2));`
    `return state;`
`}`

`bool led_update_user(led_t led_state) {`
    `rgblight_set_layer_state(0, led_state.caps_lock);`
    `return true;`
`}`

When this is added to the keymap.c then it returns:
arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   66024       0   66024   101e8 .build/planck_ez_Yatreyu.hex

Compiling: keyboards/planck/keymaps/Yatreyu/keymap.c                                               keyboards/planck/keymaps/Yatreyu/keymap.c:210:7: error: unknown type name 'rgblight_segment_t'
 const rgblight_segment_t PROGMEM my_capslock_layer[] = RGBLIGHT_LAYER_SEGMENTS(
       ^~~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/Yatreyu/keymap.c:210:56: error: implicit declaration of function 'RGBLIGHT_LAYER_SEGMENTS'; did you mean 'RGBLIGHT_LAYER_BLINK'? [-Werror=implicit-function-declaration]
 const rgblight_segment_t PROGMEM my_capslock_layer[] = RGBLIGHT_LAYER_SEGMENTS(
                                                        ^~~~~~~~~~~~~~~~~~~~~~~
                                                        RGBLIGHT_LAYER_BLINK
keyboards/planck/keymaps/Yatreyu/keymap.c:211:5: error: expected expression before '{' token
     {6, 4, HSV_RED},       // Light 4 LEDs, starting with LED 6
     ^
keyboards/planck/keymaps/Yatreyu/keymap.c:215:7: error: unknown type name 'rgblight_segment_t'
 const rgblight_segment_t PROGMEM my_layer1_layer[] = RGBLIGHT_LAYER_SEGMENTS(
       ^~~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/Yatreyu/keymap.c:216:5: error: expected expression before '{' token
     {9, 2, HSV_CYAN}
     ^
keyboards/planck/keymaps/Yatreyu/keymap.c:215:54: error: invalid initializer
 const rgblight_segment_t PROGMEM my_layer1_layer[] = RGBLIGHT_LAYER_SEGMENTS(
                                                      ^~~~~~~~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/Yatreyu/keymap.c:219:7: error: unknown type name 'rgblight_segment_t'
 const rgblight_segment_t PROGMEM my_layer2_layer[] = RGBLIGHT_LAYER_SEGMENTS(
       ^~~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/Yatreyu/keymap.c:220:5: error: expected expression before '{' token
     {11, 2, HSV_PURPLE}
     ^
keyboards/planck/keymaps/Yatreyu/keymap.c:219:54: error: invalid initializer
 const rgblight_segment_t PROGMEM my_layer2_layer[] = RGBLIGHT_LAYER_SEGMENTS(
                                                      ^~~~~~~~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/Yatreyu/keymap.c:224:7: error: unknown type name 'rgblight_segment_t'
 const rgblight_segment_t* const PROGMEM my_rgb_layers[] = RGBLIGHT_LAYERS_LIST(
       ^~~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/Yatreyu/keymap.c:224:59: error: implicit declaration of function 'RGBLIGHT_LAYERS_LIST'; did you mean 'RGBLIGHT_LAYER_BLINK'? [-Werror=implicit-function-declaration]
 const rgblight_segment_t* const PROGMEM my_rgb_layers[] = RGBLIGHT_LAYERS_LIST(
                                                           ^~~~~~~~~~~~~~~~~~~~
                                                           RGBLIGHT_LAYER_BLINK
keyboards/planck/keymaps/Yatreyu/keymap.c:224:59: error: invalid initializer
keyboards/planck/keymaps/Yatreyu/keymap.c: In function 'layer_state_set_user':
keyboards/planck/keymaps/Yatreyu/keymap.c:232:5: error: implicit declaration of function 'rgblight_set_layer_state'; did you mean 'rgblight_decrease_sat'? [-Werror=implicit-function-declaration]
     rgblight_set_layer_state(1, layer_state_cmp(state, 1));
     ^~~~~~~~~~~~~~~~~~~~~~~~
     rgblight_decrease_sat
keyboards/planck/keymaps/Yatreyu/keymap.c: At top level:
keyboards/planck/keymaps/Yatreyu/keymap.c:242:6: error: redefinition of 'keyboard_post_init_user'
 void keyboard_post_init_user(void) {
      ^~~~~~~~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/Yatreyu/keymap.c:203:6: note: previous definition of 'keyboard_post_init_user' was here
 void keyboard_post_init_user(void) {
      ^~~~~~~~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/Yatreyu/keymap.c: In function 'keyboard_post_init_user':
keyboards/planck/keymaps/Yatreyu/keymap.c:244:5: error: 'rgblight_layers' undeclared (first use in this function); did you mean 'rgblight_step'?
     rgblight_layers = my_rgb_layers;
     ^~~~~~~~~~~~~~~
     rgblight_step
keyboards/planck/keymaps/Yatreyu/keymap.c:244:5: note: each undeclared identifier is reported only once for each function it appears in
keyboards/planck/keymaps/Yatreyu/keymap.c:244:5: error: statement with no effect [-Werror=unused-value]
keyboards/planck/keymaps/Yatreyu/keymap.c: At top level:
keyboards/planck/keymaps/Yatreyu/keymap.c:283:10: error: redefinition of 'layer_state_set_user'
 uint32_t layer_state_set_user(uint32_t state) {
          ^~~~~~~~~~~~~~~~~~~~
keyboards/planck/keymaps/Yatreyu/keymap.c:230:15: note: previous definition of 'layer_state_set_user' was here
 layer_state_t layer_state_set_user(layer_state_t state) {
               ^~~~~~~~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:386: .build/obj_planck_ez_Yatreyu/keyboards/planck/keymaps/Yatreyu/keymap.o] Error 1
Make finished with errors
make: *** [Makefile:584: planck/ez:Yatreyu:flash] Error 1

ELLIOTTCABLE, 2020-08-20:
I'm having the same issue; obviously, the definitions in `quantum/rgblight.h` aren't making it to my `keymap.c`, but I have no idea why, and my C is rusty …

https://github.com/qmk/qmk_firmware/blob/83c7c66e8caab77bfb97c937c371ceb165c58102/quantum/rgblight.h#L187-L194

Full keymap: https://gist.github.com/ELLIOTTCABLE/4434d98cb349d84014fdce3121255eba
Full error: https://gist.github.com/ELLIOTTCABLE/0d2d9cba2959f377fb00863c7edba116

ELLIOTTCABLE, 2020-08-20:
Ah, okay, someone else figured it out for me: my QMK fork was out-of-date with master, and RGBLIGHT_LAYERS is a relatively new feature.

@Yatreyu, I suggest you rebase up to QMK master! Probably something along the lines of:

```sh
cd ~/Code/qmk_firmware
git fetch origin
git rebase origin/master
```

closed 2020-12-19 by stale[bot]

edz381, 2021-01-07:
@ELLIOTTCABLE I am having the same issue as - I've added the same code snippet from the QMK documentation and the compiler is complaining.  I have also run the rebase command but this has not made a difference.

**Code added**

`const rgblight_segment_t PROGMEM my_capslock_layer[] = RGBLIGHT_LAYER_SEGMENTS(
    {6, 4, HSV_RED},       // Light 4 LEDs, starting with LED 6
    {12, 4, HSV_RED}       // Light 4 LEDs, starting with LED 12
);`

**Compiler output**
`Compiling: keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c                                    keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:70:7: error: type defaults to 'int' in declaration of 'rgblight_segment_t' [-Werror=implicit-int]
 const rgblight_segment_t PROGMEM my_capslock_layer[] = RGBLIGHT_LAYER_SEGMENTS(
            ^~~~~~~~~~~~~~~~~~
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:70:34: error: expected ',' or ';' before 'my_capslock_layer'
 const rgblight_segment_t PROGMEM my_capslock_layer[] = RGBLIGHT_LAYER_SEGMENTS(
                                                                  ^~~~~~~~~~~~~~~~~
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:71:20: error: expected identifier or '(' before ',' token
     {6, 4, HSV_RED},       // Light 4 LEDs, starting with LED 6
                    ^
keyboards/kbdfans/kbd67/mkiirgb/keymaps/edz/keymap.c:73:1: error: expected identifier or '(' before ')' token
 );
 ^
cc1: all warnings being treated as errors
 [ERRORS]`

jackokring, 2021-11-24:
```
#define RGBLIGHT_SLEEP
#define RGBLIGHT_LAYERS
#define RGBLIGHT_LAYERS_OVERRIDE_RGB_OFF
#define RGBLIGHT_LAYER_BLINK
```
Or such in `config.h` ??

temocRodr, 2023-05-20:
For others who come across this issue, I had the same issue and I found that I forgot the 
`RGBLIGHT_ENABLE = yes`
within rules.mk. After adding it, compile worked for me.