qmk/qmk_firmware #10047: [Feature Request]  Defined architecture/os portability layer for QMK

cbc02009, 2020-08-15:
## Feature Request Type

- [X ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ X] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

I've found a great little board for small keyboard projects (https://www.seeedstudio.com/Seeeduino-XIAO-Arduino-Microcontroller-SAMD21-Cortex-M0+-p-4426.html) but it runs on a processor not currently supported by QMK. I've been pulling my hair out for a month trying to figure out the best way to port it, but all the code components are so tangled together and I'm having trouble separating them out.

Ideally for a project of this type you'd have several header files (gpio, usb, timer, eeprom, etc) with function prototypes that the qmk library needs to function, and then separate architectures (or OS ports) could then populate as necessary via unique port source files.  

For example the matrix scan functionality could include "gpio_port.h" which would include prototypes:
```
void set_pin_output(void);
void set_pin_input(bool pullup);
bool get_pin_level(pin_t pin);
void set_pin_level(pin_t pin, bool level);
```

which would be compiled from different source files depending on the passed architecture. 

This would allow the qmk library to be completely portable and detached from the processor it's run on. It could even detach it from the OS, so it would be easy to port it to new operating systems like RIOT.

I know that there's a protocol folder, but I couldn't find a clear definition of what needs to go into it to make a new protocol. If I missed it I apologize.

I would love to help with this, as I have a decent amount of C experience, the problem is that we use cmake/IDE for writing C at work, and so I don't have any experience whatsoever with makefiles. I'd still be happy to help in whatever way I can.

I'm ignoring the SAMD51 stuff for this discussion because it's written specifically for massdrop's boards and is not portable to other SAM chips at all. (Seriously you'd think they'd have written the "ARM-ATSAM" protocol to be compatible with more SAM chips and more boards.)

zvecr, 2020-08-15:
A fair amount of the required abstraction is already in place, be it with plans to reorg a few things to make it clearer.

For your example `matrix scan functionality could include "gpio_port.h" which would include prototypes` is covered by <https://docs.qmk.fm/#/internals_gpio_control>. We include the same matrix scanning code for currently supported AVR and ARM keyboards.

Docs on this area are lacking, though we would want super early engagement before something like a platform port happens. So in a way, this does provide a slightly useful barrier for entry.

Till someone sits down and PRs some architecture docs, the summary is something like:
* `tmk_core/common/<protocol>` - main and usb access
* `tmk_core/<platform>` - platform specific abstractions
* `drivers/<platform>` - platform specific hardware drivers
* `quantum/<feature>/<feature>_<platform>` - platform specific feature implementation

and on `develop`, we have `platforms` which is the start of the migration.

tzarc, 2020-08-15:
The mention of RIOT is a good one -- we'd already been internally discussing the development of a port to use it. The main rationale behind such an initiative is to allow for us to deprecate arm_atsam, which you've already noticed does not have a lot of flexibility. It'd also help us pick up some bluetooth support for nRF52 chips -- something that we're currently incapable of doing due to licensing of the proprietary radio stacks -- RIOT has a permissive one.

As you can probably guess from the state of arm_atsam, we'd prefer to have upstream repositories leveraged as opposed to bespoke code for an entire MCU family's HAL embedded within QMK. We can then leverage upstream development of these HALs, without incurring any major cost to QMK other than the act of performing the upgrade. As such... if you were to undertake a port to SAMD21, I'd strongly recommend targeting the integration of RIOT from the start -- it's much less likely that we dismiss the PR.

cbc02009, 2020-08-17:
That makes a lot of sense, thanks for the clarification. I'll take a look at porting for RIOT, but unfortunately I don't have any experience whatsoever with make and makefiles so it may be beyond my ability.

closed 2020-12-15 by stale[bot]

mrninhvn, 2023-11-07:
Hi, is there any update for this? Hopefully a portability layer works.

tzarc, 2023-11-07:
No.