qmk/qmk_firmware #10084: Romeo keymap wrong keycode for Enter switch?

BenRoe, 2020-08-18:
Maybe i am wrong, but shouldn't the most right switch in the second row from top, be a `KC_ENT` and not a `KC_QUOT`?
@coseyfannitutti 
https://github.com/qmk/qmk_firmware/blob/186b1b2ba2550d9a7a864a241e8f9503c37f188b/keyboards/coseyfannitutti/romeo/keymaps/default/keymap.c#L22

[image]

[image]

coseyfannitutti, 2020-08-18:
This was an intentional choice on my part, and maybe I should have been more clear with my reasoning on the project's page. When I was testing the layout, I found myself constantly hitting enter and cutting my sentences off when chatting in Discord, for example, and trying to hit apostrophe. I found that placing enter on layer 1 made striking enter more intentional.

Because it's a 40% with a mostly standard staggered layout, I expected a large percentage of users to be new to 40% and/or unconventional layouts. I think it could help touch typists not used to a compact layout to be able to mostly type naturally without accidentally submitting forms or chat messages.

I figured if anyone didn't like it, they could easily rebind it in VIA.

BenRoe, 2020-08-18:
Thank you for the detailed explanation.
It make sense.

closed 2020-08-18 by BenRoe