qmk/qmk_firmware #10093: [Feature Request] KBD67RGB mkii v2 ansi 68 key layout

itsmacca97, 2020-08-19:
Could you please add in a 68 key ansi layout with three 1u keys to the left of the arrow keys (alt, MO1, Ctrl) for the KBD Fans KBD67 mkii v2 RGB

## Feature Request Type
[image]

- [ y] Core functionality
- [n ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [n ] Alteration (enhancement/optimization) of existing feature(s)
- [n ] New behavior

## Description

Could you please add in a 68 key ansi layout with three 1u keys to the left of the arrow keys (alt, MO1, Ctrl) for the KBD Fans KBD67 mkii/' v2 RGB

mrsendyyk, 2020-08-19:
Did you mean added to [QMK Configurator](https://config.qmk.fm/#/kbdfans/kbd67/mkiirgb/v2/LAYOUT_65_ansi_blocker)?

itsmacca97, 2020-08-19:
I would like the above layout added to the kbd fans kbd67 mkii RGB V2 on QMK Configurator yes please. I accidentally flashed the wrong board and can only reflash with the template on there and lose functionality of one of my key. 
Thanks 😊😊

mrsendyyk, 2020-08-20:
Ok, maybe i can help you after fixed my Issues [#794](https://github.com/qmk/qmk_configurator/issues/794) on  [qmk/qmk_configurator](https://github.com/qmk/qmk_configurator).

itsmacca97, 2020-08-24:
> Hi @itsmacca97, i think KBD67 v2 MKII RGB PCB not support `LAYOUT_65_ansi`. Based on picture from [KBDfans](https://kbdfans.com/products/kbd67-mkii-pcb), that pcb only support `LAYOUT_65_ansi_blocker`.
> 
> [image]
> 
> [image]
> 
> @mechmerlin & @moyi4681 as [KBD67 Maintainer](https://github.com/qmk/qmk_firmware/tree/master/keyboards/kbdfans/kbd67), is that correct?
> 
> Also @fauxpark, can you help me again? Sorry...

I'm an idiot. I just realised that I got the DZ65 and got confused not seeing it there and thinking I had the kbd67 smh

mrsendyyk, 2020-08-25:
> I'm an idiot. I just realised that I got the DZ65 and got confused not seeing it there and thinking I had the kbd67 smh

[QMK Configurator](https://config.qmk.fm/#/dztech/dz65rgb/v2/LAYOUT_65_ansi)

itsmacca97, 2020-08-25:
> [QMK Configurator](https://config.qmk.fm/#/dztech/dz65rgb/v2/LAYOUT_65_ansi)

As you can very much tell I am a complete NOOB when it comes to this stuff. So much research in a short period had my mind all over the place. Thanks for dealing with my incompetence. Much appreciated (:

robotesque, 2020-11-11:
Hey,
I'm actually having the exact issue the OP described. I do have a KDB67RGB mkii v2 and the only option for it I see in QMK Configurator is the 67 key layout, but I need a 68 key layout like this:

[image]

I'm currently using the kbdfans/kbd67/mkiirgb/v2 template in the configurator and everything works but I'm missing 1 key. The kbdfans/kbd67/rev2 has the correct layout I need but it doesn't work properly with my pcb.

[image]

Thank you for the support!

closed 2021-03-16 by stale[bot]