qmk/qmk_firmware #10096: [Bug] Using only right side encoder in split keyboard

karnadii, 2020-08-19:
## Describe the Bug
I have this handwired keyboard that use only the right side of encoder
[image]

defining only the right side of the encoder throw me error that ENCODERS_PAD_A is undeclared
```
#define ENCODERS_PAD_A_RIGHT { D1 }
#define ENCODERS_PAD_B_RIGHT { D0 }
```
```
Compiling: quantum/split_common/transport.c                                                        quantum/split_common/transport.c:20:31: error: 'ENCODERS_PAD_A' undeclared here (not in a function); did you mean 'ENCODER_ENABLE'?
 static pin_t encoders_pad[] = ENCODERS_PAD_A;
                               ^~~~~~~~~~~~~~
                               ENCODER_ENABLE
```
i have to declare ENCODERS_PAD_A too

```
// Right encoder
// I can not use only right encoder, I have to define left encoder as well
#define ENCODERS_PAD_A { D1 } // dummy encoder to passs error. there is no left encoder.
#define ENCODERS_PAD_B { D0 } // dummy encoder to passs error. there is no left encoder.
#define ENCODERS_PAD_A_RIGHT { D1 }
#define ENCODERS_PAD_B_RIGHT { D0 }
#define ENCODER_RESOLUTION 4
```
the left side D1 and D0 is used for oled. and the right side is used for the encoder.

## System Information

 - Keyboard:  qsplit (my own handwired keyboard)
   - Revision (if applicable):  
 - Operating system: Macos
 - AVR GCC version: 8.4.0

 - ARM GCC version: 8.3.1

 - QMK Firmware version: 0.9.46

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

drashna, 2020-12-01:
The "right" define is for when the right side encoder is different from the left side. (such as flipped, or wired up differently). 

You want to use just the normal defines for the pads.

karnadii, 2020-12-02:
The master is on the left with the oled, if the master is on the right sure no need to declare the right encoder.

karnadii, 2020-12-02:
My point is if you have split keyboard and using encoder, don't force both side to have encoder. If I want to use only one encoder in my split, don't throw error.

BalzGuenat, 2021-03-01:
I have just encountered this limitation myself. My keyboard has the master side left and an encoder on the right half. Like @karnadii's board, mine is not symmetric and has no encoder pads on the left side. The current encoder code makes some assumptions about the halves being symmetric, at least in number of encoder pads.

I also had to resort to using some unused pins as a dummy encoder for left.

I will see if I can make some changes to `encoder.c` to make this work.

zvecr, 2022-08-28:
Fixed by https://github.com/qmk/qmk_firmware/pull/12090

closed 2022-08-28 by zvecr