qmk/qmk_firmware #10148: [Bug] dz60rgb/v1 fails to compile: "No rule to make target [...] lib/printf/printf.o"

antoineco, 2020-08-23:
## Describe the Bug

All keymaps of the `dztech/dz60rgb/v1` keyboard fail to compile with the following error:

```console
$ make dztech/dz60rgb/v1:default
Compiling: keyboards/dztech/dz60rgb/dz60rgb.c                                                       [OK]
Compiling: keyboards/dztech/dz60rgb/keymaps/default/keymap.c                                        [OK]
[...]
make[1]: *** No rule to make target '.build/obj_dztech_dz60rgb_v1_default/lib/printf/printf.o', needed by '.build/dztech_dz60rgb_v1_default.elf'.  Stop.
Make finished with errors
make: *** [Makefile:584: dztech/dz60rgb/v1:default] Error 1
```

`v2` of the same keyboard builds just fine.

## System Information

 - Keyboard:  `dztech/dz60rgb`
   - Revision:  `v1`
 - Operating system: `Ubuntu 20.04.1 LTS`
 - AVR GCC version: `5.4.0`
 - ARM GCC version: `(15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]`
 - QMK Firmware version: `0.9.55` (tried previous versions down to 0.9.46, same issue)
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

antoineco, 2020-08-23:
It compiles on Debian 10, so it might be some missing dependency but `qmk setup` doesn't complain.

chipperdoodles, 2020-09-03:
Had the same issue with a different board, ran `git submodule update` and seemed to have fixed it

antoineco, 2020-09-07:
@chipperdoodles nice catch! My submodules were indeed outdated:

```
Cloning into '/git/qmk/qmk_firmware/lib/printf'...
Cloning into '/git/qmk/qmk_firmware/lib/vusb'...
Submodule path 'lib/printf': checked out 'd3b984684bb8a8bdc48cc7a1abecb93ce59bbe3e'
Submodule path 'lib/vusb': checked out '9a42d205eb60faca494ff4eabce8d59f0ec0fc7f'
```

Now the firmware builds like a charm 👍 
Closing

closed 2020-09-07 by antoineco