qmk/qmk_firmware #10163: Insufficient information on Leader key page

CrisHale, 2020-08-25:
There is a function LEADER_EXTERNS() which isn't explained. It also doesn't say where it should go. I haven't seen anything that mentioned whether keymap.c needs to have any particular order to it, but I would assume it does.
Can anybody answer this and update the documentation to reflect it?

Thanks,
Cris

ridingqwerty, 2020-09-11:
The guts of these macros is in `quantum/process_keycode/process_leader.h`:

```
#define LEADER_EXTERNS()                \
    extern bool     leading;            \
    extern uint16_t leader_time;        \
    extern uint16_t leader_sequence[5]; \
    extern uint8_t  leader_sequence_size
#define LEADER_DICTIONARY() if (leading && timer_elapsed(leader_time) > LEADER_TIMEOUT)
```

`LEADER_EXTERNS()` is simply setting up the timer and sequence variables used by the leader feature.  It needs to be declared before the use of leader stuff in `matrix_scan_user` in a global scope.  This could be in a few places (especially if you are doing separate compilation by declaring distinct files at the keymap-level or in userspace) but the `keymap.c` is generally fine.

The `leading = false` bit is indicating that process the leader sequence is done, so that the `SEQ` macros can check for the proper keys and execute the appropriate action.

For the most part you don't really need to know the internals of these functions, but if you're curious `quantum/process_keycode/process_leader.{c,h}` has that information.

CrisHale, 2020-09-11:
Thanks, so if the `leading = false` part weren't used, the function wouldn't work at all, then?

ridingqwerty, 2020-09-11:
I'm not entirely sure what would happen.  The `process_leader` function here

```
bool process_leader(uint16_t keycode, keyrecord_t *record) {
    // Leader key set-up
    if (record->event.pressed) {
        if (leading) {
            if (timer_elapsed(leader_time) < LEADER_TIMEOUT) {
#    ifndef LEADER_KEY_STRICT_KEY_PROCESSING
                if ((keycode >= QK_MOD_TAP && keycode <= QK_MOD_TAP_MAX) || (keycode >= QK_LAYER_TAP && keycode <= QK_LAYER_TAP_MAX)) {
                    keycode = keycode & 0xFF;
                }
#    endif  // LEADER_KEY_STRICT_KEY_PROCESSING
                if (leader_sequence_size < (sizeof(leader_sequence) / sizeof(leader_sequence[0]))) {
                    leader_sequence[leader_sequence_size] = keycode;
                    leader_sequence_size++;
                } else {
                    leading = false;
                    leader_end();
                }
#    ifdef LEADER_PER_KEY_TIMING
                leader_time = timer_read();
#    endif
                return false;
            }
        } else {
            if (keycode == KC_LEAD) {
                qk_leader_start();
            }
        }
    }
    return true;
}
```

will set `leading = false` if the max number of sequence keys (5) is reached, so I suspect five key sequences would still work just fine.  I do wonder if closing out the `if (timer_elapsed(leader_time) < LEADER_TIMEOUT) {` block with an `else` case that sets `leading = false` would obviate the need to do that in `matrix_scan_user`.

ridingqwerty, 2020-09-11:
Actually that might be Bad.  Best to just follow the format in the doc.

closed 2021-01-10 by stale[bot]