qmk/qmk_firmware #10165: Quantum keycode RESET throwing error

CrisHale, 2020-08-25:
I thought I'd try a leader sequence for resetting the keyboard, but when it came to compiling I got the error "unsigned conversion from 'int' to 'uint_8t'.
I used the line "register_code(RESET);"
Is there something specific that needs to be done to use these quantum keycodes, and if that is the issue, could that be added to the documentation page?

Thanks,
Cris

zvecr, 2020-08-25:
You cannot use the keycodes like that, use `reset_keyboard()` directly instead.

CrisHale, 2020-08-25:
Great, thank you. Perhaps a reference to that function could be included on the quantum keycodes page to help people like myself who learn about it from that page?

closed 2020-12-24 by stale[bot]