qmk/qmk_firmware #10198: Can't use alternative layout feature going by documentation

CrisHale, 2020-08-29:
The [reference keymap extras](https://docs.qmk.fm/#/reference_keymap_extras?id=sendstring-support) page says that including
`#include "sendstring_keymap_uk.h"`
will allow the SENDSTRING function to send strings as with the UK layout, but it says no such file or directory is found. Is the documentation out of date?

Thanks,
Cris

fauxpark, 2020-08-29:
> If you are using a different layout, you can also #include "sendstring_*.h" (as above) in your keymap to override the lookup tables used for mapping ASCII characters to keycodes.

It's `sendstring_uk.h`.

CrisHale, 2020-08-29:
Thanks. So the documentation if wrong? Is there someone I should inform?

fauxpark, 2020-08-29:
The documentation is correct. The `*` here stands in for the language suffixes, not the whole thing. You have `keymap_uk.h` and `sendstring_uk.h`.

CrisHale, 2020-08-30:
Well it doesn't specify that - or even imply it - in the wording on the page. I've got my answer, so I'm sorted. If you want to make it difficult for others in the future I'm not going to stop you.

closed 2020-08-30 by CrisHale