qmk/qmk_firmware #10199: [MIDI] Sustain effect when same notes are played.

3araht, 2020-08-29:
I referred [here](https://github.com/qmk/qmk_firmware/blob/9b232a7f885e320061aae994713acd31297eac5c/keyboards/sentraq/s60_x/keymaps/bluebear/keymap.c#L664) to assign a chord to each keys. When multiple chords are played and when the chords share the same note(s), it activates a sustain effect that stays until you unplug the keyboard.
When you are using GarageBand, the symptom can be fixed by clicking "Revert" button on the bottom left corner on the GarageBand window. But want to avoid it from the first place...
Could someone help me out?

Thank you in advance!

3araht, 2020-09-11:
Thanks! I appreciate it.

jakobkg, 2020-09-12:
Hmm, I've been trying for a while now but I'm unable to replicate the issue. I've set two keys as `MI_CH_C`  and `MI_CH_CDom7` and playing them both at the same time just plays the two chords. When I release one chord, the notes that are shared are all turned off.

I'm flashing QMK on a pro micro and connecting it to my PC which is running Ableton Live 10.

Example code:
```
#include QMK_KEYBOARD_H
#include "analog.h"
#include "qmk_midi.h"

enum custom_keys {
  // MIDI Chord Keycodes - Major

  MI_CH_C,
  MI_CH_Cs,
  MI_CH_Db = MI_CH_Cs,
  MI_CH_D,
  MI_CH_Ds,
  MI_CH_Eb = MI_CH_Ds,
  MI_CH_E,
  MI_CH_F,
  MI_CH_Fs,
  MI_CH_Gb = MI_CH_Fs,
  MI_CH_G ,
  MI_CH_Gs,
  MI_CH_Ab = MI_CH_Gs,
  MI_CH_A,
  MI_CH_As,
  MI_CH_Bb = MI_CH_As,
  MI_CH_B,

  //MIDI Chord Keycodes Dominant Seventh

  MI_CH_CDom7,
  MI_CH_CsDom7,
  MI_CH_DbDom7 = MI_CH_CsDom7,
  MI_CH_DDom7,
  MI_CH_DsDom7,
  MI_CH_EbDom7 = MI_CH_DsDom7,
  MI_CH_EDom7,
  MI_CH_FDom7,
  MI_CH_FsDom7,
  MI_CH_GbDom7 = MI_CH_FsDom7,
  MI_CH_GDom7,
  MI_CH_GsDom7,
  MI_CH_AbDom7 = MI_CH_GsDom7,
  MI_CH_ADom7,
  MI_CH_AsDom7,
  MI_CH_BbDom7 = MI_CH_AsDom7,
  MI_CH_BDom7
};

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
    /* Base */
    [0] = LAYOUT(
    MI_CH_C, MI_CH_CDom7
    )
};

bool process_record_user (uint16_t keycode, keyrecord_t *record) {

  uint16_t root_note = MIDI_INVALID_NOTE; // Starting value for the root note of each chord

  switch(keycode) {

	// MIDI Chord Keycodes

  case MI_CH_C ... MI_CH_B: // Major Chords
	root_note = keycode - MI_CH_C + MI_C;
	process_midi(root_note, record);
	process_midi(root_note + 4, record); // Major Third Note
	process_midi(root_note + 7, record); // Fifth Note
	break;

  case MI_CH_CDom7 ... MI_CH_BDom7: // Dominant 7th Chord
	root_note = keycode - MI_CH_CDom7 + MI_C;
	process_midi(root_note, record);
	process_midi(root_note + 4, record); // Major Third Note
	process_midi(root_note + 10, record); // Minor Seventh Note
	break;
  }
  return true;
}
```

For example:
I press `MI_CH_C` so the notes C, E and G play. I keep that held and press `MI_CH_CDom7`. `MI_CH_CDom7` plays the notes C, E and Bb so the C and E are re-triggered and a Bb is played, while the G keeps ringing.
I then release `MI_CH_C`, and the notes C, E and G are all stopped. This means that the only note still playing is Bb even though I'm still holding `MI_CH_CDom7`.
The opposite is also true, so if I release `MI_CH_CDom7` the C, E and Bb stop and the only note still playing is a G.

This is an audio recording of both these tests:
https://vocaroo.com/23kuNYatIeK

I also tested something equivalent by setting two keys to `MI_C`, and the same thing happens. If I press them both and release only one of the keys, the note stops since the note off message gets sent when a key is released even though I'm still holding the note on another key.

This is kind of bad in my opinion, but it's a different problem to the one you're describing. Maybe this is a problem specific to GarageBand? I'd love it if you could provide more details about your setup and if the problem persists with a more barebones firmware or in otgher apps than GarageBand.

3araht, 2020-09-12:
I see. That is also kind of strange.
At least, we share an issue that the note which is pressed stops ringing when a duplicate note is released at your end.
On my side, maybe related to GarageBand, the note which has to stop keeps ringing.

I'll try "Ableton Live 10 Lite" and see what happens on my side.

I'm having trouble recording sounds with vocaroo now. I'll share it when I solved it.

I'm using Pro Micro, too. The keyboard I use is shown here:
https://github.com/3araht/giabalanai

Thanks!

jakobkg, 2020-09-12:
I recorded a file on my PC and uploaded to vocaroo with the little "upload" button on the page instead of recording directly on vocaroo, that should work for you too.

That is a very interesting board you're making, I like it a lot!

3araht, 2020-09-12:
LOL. I should have done that.
Here you are. Now I'm 40s now.
https://voca.ro/14nwiestv17w

I'm glad you like the board.

3araht, 2020-09-12:
Yep, the symptom seems MIDI software dependent.
I was able to replicate yours. Sorry for using a microphone again. didn't have the authority to export data directly from Ableton Live 10 Lite.
https://voca.ro/18F45DA201XJ

The last sound (A#) was tapped while I keep pressing Cdom7 code button.
All sounds stop ringing.

Compared to the sustain effect seen with GarageBand, the symptom seen with Ableton Live 10 would be easier to live with, in my opinion.

Well, a lot of novice players, including me, use GarageBand, so I wish if there is any solution to deal with it.

3araht, 2020-09-13:
Logic Pro also shows the same symptom.
Do you think it is possible to fix both symptoms?

jakobkg, 2020-09-17:
How strange! I've been busy with work, but I'll have some time today to install Logic and do some testing. I think I have an idea for a fix, but it'll take me a bit to test and implement.

3araht, 2020-09-18:
It seems I accidentally found a fix!!!
See below for the modification I made.
This solved the sustain problem, but not the new issue you found with Ableton Live 10...
However, it is much much better for GarageBand users and Logic Pro users.
Is this modification acceptable?

```

diff --git a/quantum/process_keycode/process_midi.c b/quantum/process_keycode/process_midi.c
index b2fb902eb..e52577014 100644
--- a/quantum/process_keycode/process_midi.c
+++ b/quantum/process_keycode/process_midi.c
@@ -68,10 +68,12 @@ bool process_midi(uint16_t keycode, keyrecord_t *record) {
             uint8_t tone     = keycode - MIDI_TONE_MIN;
             uint8_t velocity = compute_velocity(midi_config.velocity);
             if (record->event.pressed) {
-                uint8_t note = midi_compute_note(keycode);
-                midi_send_noteon(&midi_device, channel, note, velocity);
-                dprintf("midi noteon channel:%d note:%d velocity:%d\n", channel, note, velocity);
-                tone_status[tone] = note;
+                if (tone_status[tone] == MIDI_INVALID_NOTE) {
+                    uint8_t note = midi_compute_note(keycode);
+                    midi_send_noteon(&midi_device, channel, note, velocity);
+                    dprintf("midi noteon channel:%d note:%d velocity:%d\n", channel, note, velocity);
+                    tone_status[tone] = note;
+                }
             } else {
                 uint8_t note = tone_status[tone];
                 if (note != MIDI_INVALID_NOTE) {
```

3araht, 2020-09-18:
I got a question from ralfkaa, what I intended to do. I apologize for not describing it beforehand.
See below. 
https://github.com/qmk/qmk_firmware/issues/9456#issuecomment-694824972

jakobkg, 2020-09-19:
Very clever, nice find! I feel like the three of us have slightly different ideas about how pressing the same note multiple times should be handled, it's probably best if we keep it to discussing in the PR you've opened so we don't end up having the same conversation in multiple different places!

jakobkg, 2020-09-19:
Actually, here are my thoughts before I start coding something that might be unneccesary or not work

I would personally like it if it was possible to send note on messages for the same note multiple times, and as long as the note off message is only sent once per note there shouldn't be any issues, at least from my very quick read-through of the MIDI note message specification. I'm fairly certain the issues you've been experiencing have been due to software receiving note off messages for notes that are already off, but I'll have to test a little to be sure of that.

My thinking is that it should be possible to use a QMK MIDI device to produce the effect of playing the same not on different strings on a guitar: every time you pluck a string (press a key) a new note is played (a new MIDI note on message is sent) but if you mute one of the strings (release one of the keys) the note doesn't stop as long as it's still playing on a different string (a different key with the same note is still held).

A very rough sketch of this idea would be 
```
when note key pressed:
  add note to array of currently playing notes
  send midi note on message

when note key released:
  remove note from array of currently playing notes
  if note is not in array of currently playing notes:
    send midi note off message
```

So if I for example have two keys mapped to `MI_C_1` and I press the first one, the contents of the array would be `[ MI_C_1 ]` and the note would trigger. If I then press the other `MI_C_1` key, the note would be triggered again and the contents of the array would be `[MI_C_1, MI_C_1]`. If I then release one of the keys, the array would have one element removed and only be `[ MI_C_1 ]`. Even though I released one key, another key with the same note is still pressed and so the note off message isn't sent. If I then release the other `MI_C_1` key, the array would be empty and the MIDI note off message would be sent.

I _hope_ QMK already keeps a list of currently pressed keys somewhere to make this easier, but I haven't been able to find anything yet. If done from scratch this might end up a little memory hungry, but unless someone made a keyboard with lots of the same note it likely won't be an issue.

3araht, 2020-09-19:
How nice! I like your idea. Your solution is "how it should be". It will solve the issue you find with Ableton Live 10, and what we share now with GarageBand, Logic Pro X, and so forth, after the fix shown in the PR I sent. 
So I look forward to it.
The PR I sent is somewhat a temporal update. But it makes GarageBand and Logic Pro X users' life easier at the moment.

Regarding how many same notes on a keyboard, I'm not sure how many out there. 
For example, there are 60 keys for chords as you can see [here](https://github.com/3araht/giabalanai/blob/master/pictures/giabalanai_overview.jpg) on the left hand side keyboard.
I have no idea which has duplicate notes... At least 5 maybe? No one should play all of them at the same time though. But it has to be prepared for such situation.

3araht, 2020-10-04:
The fix has been made and merged. See here for details. 
https://github.com/qmk/qmk_firmware/pull/10361

closed 2020-10-04 by 3araht

jakobkg, 2020-10-06:
Great work! I have been very busy lately and have not been taking the time or energy to work on QMK things so it's great that you've picked this up and solved it!

3araht, 2020-10-06:
Oh, but the issue you found remains there.
It's just the quick fix for the sustain effect issue.