qmk/qmk_firmware #10209: QMK logo for Wikipedia/Wikicommons

nunmap, 2020-08-31:
Hello,
Could you please publish the QMK logo as a svg & and release it under one of the following licenses please?

* Creative Commons Attribution ShareAlike 4.0 (legal code)
* Creative Commons Attribution ShareAlike 3.0 (legal code)
* Creative Commons Attribution ShareAlike 2.5 (legal code)
* Creative Commons Attribution 4.0 (legal code)
* Creative Commons Attribution 3.0 (legal code)
* Creative Commons Attribution 2.5 (legal code)
* Creative Commons CC0 Waiver (release all rights, like public domain: legal code)

So the logo can freely be used on Wikipedia articles about QMK.

fauxpark, 2020-11-29:
@qmk/directors ?

tylert, 2021-06-16:
There are SVG and PNG logos at https://github.com/qmk/qmk.fm/tree/gh-pages/assets/images (in the "qmk/qmk.fm" repo) and https://qmk.fm/powered/ wuich might be suitable as a basis for further logo work.  I'm sure some additional out-of-band discusion surrounding their use (perhaps on Discord) would also be required.

closed 2022-08-29 by zvecr