qmk/qmk_firmware #10230: Redox (wired) - Flashing the keyboard with a recently compiled firmware leaves the slave side unresponsive

CTocha, 2020-09-03:
## Describe the Bug

Flashing the keyboard with a recently compiled firmware (default layout) leaves the slave side unresponsive. If using saved layouts (both the default and custom ones) from a month ago, it works without problems.

## System Information

 - Keyboard:  Redox
   - Revision (if applicable):  
 - Operating system: Windows
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version: 

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

piscue, 2020-11-07:
did you tried to compile locally on your machine? does the resulting files differ?

CTocha, 2020-11-10:
I don't know how to compile it locally. The new file I compiled online differs from the old one, though, although the keyboard layout is the same.

fauxpark, 2020-11-10:
Are you flashing both sides?

CTocha, 2020-11-10:
Not sure if I understand the question. I didn't need to flash them separately in the past, as they're always connected and set up accordingly.

fauxpark, 2020-11-10:
As I understand it, the Redox uses two Pro Micros, one on each side - but flashing one does not flash the other, so when changes to the split protocol occur, you must flash both sides (we try to keep these changes to a minimum, for this exact reason).

CTocha, 2020-11-11:
Thanks a lot, that did worked perfectly!

closed 2020-11-11 by CTocha