qmk/qmk_firmware #10270: Mac OS Problems Detected, "Can't run `bin/qmk --version`"

semaj4712, 2020-09-08:
Trying to get QMK to install via homebrew on mac, I got everything working I think except I keep getting this error.  Can't run `bin/qmk --version` 

How do I go about fixing this, other posts have suggested adding linking python3.8 to homebrew which I already did, so I have no idea what to try next.

semaj4712, 2020-09-08:
I got it working, I installed python from the website instead of homebrew and then it worked after installing dependencies

leonardlorenz, 2020-10-19:
Reminder to close your issue after you solved it.

baffalop, 2021-01-20:
I have the same issue. Running `qmk --version` myself outputs `unknown`. I tried installing python using the installer downloaded from the website, but this didn't change anything. I also tried uninstalling my homebrew version (having to do `brew uninstall --ignore-dependencies python`), but this led to `qmk` not working at all, as its homebrew installation was linked to my python homebrew installation.

Even if installing python a different way fixes this issue for most users, I don't think that resolves the issue. Qmk shouldn't require a very specific python installation, especially a non-homebrew one given that `qmk` itself is installed via homebrew, and the setup step that automatically installs dependencies _also_ uses homebrew.

Stats:
OS: macOS 10.15.7
Exact setup command run: `qmk setup -H ~/Developer/qmk_firmware baffalop/qmk_firmware`
Python version on homebrew: `python@3.9: stable 3.9.1 (bottled)`

baffalop, 2021-01-20:
Although, I just tried running an example `qmk compile` and everything seems to have worked. Is the `qmk --version` just a final verification step in `setup`?

yannickbuntsma, 2021-02-24:
I spent a couple hours yesterday trying to get this running. I was able to `flash` a JSON file, but not `compile`. Had a look in the QMK source code as to where the "Can not run `bin/qmk`!" warning I saw was coming from. That's here:
https://github.com/qmk/qmk_firmware/blob/c66df1664497546f32662409778731143e45a552/message.mk#L89-L90

Then looked at what triggers that message, which is here:
https://github.com/qmk/qmk_firmware/blob/5edfdeff50e1741a93dae9154fa413c3a7752d32/Makefile#L504-L505

Ran the `bin/qmk hello` command, which gave me some very useful information:
```
Could not find module appdirs!
Please run `python3 -m pip install -r /Users/yannick/qmk_firmware/requirements.txt` to install required python dependencies.
```

Running that command installed all missing dependencies and solved my problem!

Nebrethar, 2021-03-24:
I am running WSL2 with Ubuntu 20.04 (I got it to work)

The above solution did not work for me, but it looks like `bin/qmk` was using windows-style line breaks. Here is the error from the first line in the file:

```
/usr/bin/env: ‘python3\r’: No such file or directory
```

I used dos2unix [as outlined here](https://askubuntu.com/questions/896860/usr-bin-env-python3-r-no-such-file-or-directory) and it worked for me.

I am glad to create a PR but I really only changed line breaks 😁 I warn that this may be indicative of a larger compatibility issue with the files in this repo.

yiancar, 2021-03-27:
Above worked for me as well Please PR :)))

inket, 2021-05-23:
Ran into the same issue on macOS 11.3 with Apple Silicon.
2 homebrew installations, one for Apple Silicon arm64 and one for Intel x86_64.

QMK installed with the x86_64 homebrew, no issues during the install.
`qmk setup` complained about missing `arm-none-eabi-gcc` and `avr-gcc`, and installed them automatically using homebrew.

`qmk setup` then displayed "Ψ QMK is ready to go"

Weirdly enough, it still couldn't compile despite other commands working fine
```sh
$ qmk --version
unknown

$ qmk hello
Ψ Hello, World!

$ qmk compile -kb kbdfans/kbd67/rev2 -km finalISO
Ψ Compiling keymap with make -j 1 kbdfans/kbd67/rev2:finalISO

QMK Firmware 0.12.50
ERROR: Can not run bin/qmk!

 Please run util/qmk_install.sh to install all the dependencies QMK requires.

make: *** [kbdfans/kbd67/rev2:finalISO] Error 1
```

The fix was to do
```sh
export PATH="/usr/local/opt/avr-gcc@8/bin:$PATH"
export PATH="/usr/local/opt/arm-gcc-bin@8/bin:$PATH"
export PATH="/usr/local/Cellar/qmk/0.0.45/libexec/bin/:$PATH"
```

Then `qmk compile` ran fine! `qmk` just needed its own Python installation to be in the PATH to pick it up.

laughedelic, 2021-05-24:
I have exactly the same problem as @inket ☝️ on macOS 11.2.3 (MacBook Pro late 2013).
Adding those paths to the env helped.

The error message is confusing:

> Please run util/qmk_install.sh to install all the dependencies QMK requires.

Running `util/qmk_install.sh` upgrades **all** (unrelated) brew installations. I wanted to run it several times and stopped, because it seems completely unnecessary and takes a long time. I ran other commands in the script:

```
brew link --overwrite arm-gcc-bin@8
brew link --force --overwrite arm-gcc-bin@8
python3 -m pip install -r requirements.txt
```

but it didn't help. In the end I did a full brew upgrade, just to exclude the possibility that it is actually needed. It didn't help (as expected).

fauxpark, 2021-05-24:
> The fix was to do
> 
> ```shell
> export PATH="/usr/local/opt/avr-gcc@8/bin:$PATH"
> export PATH="/usr/local/opt/arm-gcc-bin@8/bin:$PATH"
> export PATH="/usr/local/Cellar/qmk/0.0.45/libexec/bin/:$PATH"
> ```
> 
> Then `qmk compile` ran fine! `qmk` just needed its own Python installation to be in the PATH to pick it up.

Do not do this, you will have problems when these packages are eventually upgraded. And you shouldn't need to manually add things to `PATH`, anyway.

fauxpark, 2021-05-24:
> I am glad to create a PR but I really only changed line breaks 😁 I warn that this may be indicative of a larger compatibility issue with the files in this repo.

There is no PR needed here - `bin/qmk` does not contain any Windows-style line breaks. This is likely something you or you Git has done.

inket, 2021-05-24:
> And you shouldn't need to manually add things to PATH, anyway.

Exactly, so something is wrong with qmk.

My suggested solution works. Obviously it's a temporary solution, but it works. People would also (obviously) need to make sure the version number is correct, and I'll leave that to them.

I'm glad to be able to share that solution so that people who have the issue can move on with life instead of fighting their tools  :)

fauxpark, 2021-05-25:
> Exactly, so something is wrong with qmk.

No, something is wrong with your brew setup - it works perfectly fine for me. Most likely it is due to the M1 stuff; there is a whole extra process required to get Homebrew running properly in Rosetta.

dievartp, 2021-05-25:
> Ran into the same issue on macOS 11.3 with Apple Silicon.
> 2 homebrew installations, one for Apple Silicon arm64 and one for Intel x86_64.
> 
> QMK installed with the x86_64 homebrew, no issues during the install.
> `qmk setup` complained about missing `arm-none-eabi-gcc` and `avr-gcc`, and installed them automatically using homebrew.
> 
> `qmk setup` then displayed "Ψ QMK is ready to go"
> 
> Weirdly enough, it still couldn't compile despite other commands working fine
> 
> ```shell
> $ qmk --version
> unknown
> 
> 
> $ qmk hello
> Ψ Hello, World!
> 
> 
> $ qmk compile -kb kbdfans/kbd67/rev2 -km finalISO
> Ψ Compiling keymap with make -j 1 kbdfans/kbd67/rev2:finalISO
> 
> 
> QMK Firmware 0.12.50
> ERROR: Can not run bin/qmk!
> 
>  Please run util/qmk_install.sh to install all the dependencies QMK requires.
> 
> make: *** [kbdfans/kbd67/rev2:finalISO] Error 1
> ```
> 
> The fix was to do
> 
> ```shell
> export PATH="/usr/local/opt/avr-gcc@8/bin:$PATH"
> export PATH="/usr/local/opt/arm-gcc-bin@8/bin:$PATH"
> export PATH="/usr/local/Cellar/qmk/0.0.45/libexec/bin/:$PATH"
> ```
> 
> Then `qmk compile` ran fine! `qmk` just needed its own Python installation to be in the PATH to pick it up.

I had the same problem on my mac - turned out there was an issue with 'halo' which was missing from python 
$ pip install halo 
did the trick - it all works fine now

garyng, 2021-06-20:
> 
> 
> > I am glad to create a PR but I really only changed line breaks 😁 I warn that this may be indicative of a larger compatibility issue with the files in this repo.
> 
> There is no PR needed here - `bin/qmk` does not contain any Windows-style line breaks. This is likely something you or your Git has done.

I am not very sure, but seems like `.gitattributes` in the repo causes all the file to have `CRLF` on Windows machine.

fauxpark, 2021-06-21:
I'm also on Windows and I don't see this issue, although I do have `core.eol=lf` in my .gitignore.

garyng, 2021-06-21:
You meant `.gitconfig`? I have `autocrlf = false` in mine, not sure whether this causes the issue.

fauxpark, 2021-06-21:
Whoops, yeah, and I have that too.

garyng, 2021-06-21:
Uhmmm, maybe I'm missing the `core.eol` config.

closed 2021-09-19 by skullydazed