qmk/qmk_firmware #10286: [Bug] Flashing with "qmk flash" broken on Ubuntu 20.04 LTS

mkautto, 2020-09-10:
After #10070 flashing with "qmk_flash" is broken on Ubuntu 20.04 LTS and possibly other distros too. The version of dfu-programmer in the Ubuntu repos is 0.6.1 and it does not support the --force flag. The command exits without flashing and the keyboard is left in bootloader mode.

Frenzie, 2020-09-11:
Yup, I just spent close to half an hour figuring this out. First time with QMK. :-(

It looks like that version is slightly ancient (but running the commands without `--force` works fine for me):
https://github.com/dfu-programmer/dfu-programmer/releases

Frenzie, 2020-09-11:
The way it's currently structured it should be written something like this (but I think a greater than 0.7 might be more future-proof than contains 0.7):

```diff
diff --git a/tmk_core/avr.mk b/tmk_core/avr.mk
index 3cb34dc66..336a83e9d 100644
--- a/tmk_core/avr.mk
+++ b/tmk_core/avr.mk
@@ -122,13 +122,14 @@ define EXEC_DFU
                if [ "$(1)" ]; then \
                        $(DFU_PROGRAMMER) $(MCU) flash --force --eeprom $(QUANTUM_PATH)/split_common/$(1);\
                fi; \
+               $(DFU_PROGRAMMER) $(MCU) flash --force $(BUILD_DIR)/$(TARGET).hex;\
        else \
                $(DFU_PROGRAMMER) $(MCU) erase; \
                if [ "$(1)" ]; then \
                        $(DFU_PROGRAMMER) $(MCU) flash-eeprom $(QUANTUM_PATH)/split_common/$(1);\
                fi; \
+               $(DFU_PROGRAMMER) $(MCU) flash $(BUILD_DIR)/$(TARGET).hex;\
        fi; \
-       $(DFU_PROGRAMMER) $(MCU) flash --force $(BUILD_DIR)/$(TARGET).hex;\
        $(DFU_PROGRAMMER) $(MCU) reset
 endef
```

rebasar, 2020-09-12:
Having the same problem with Debian Buster.

closed 2020-09-13 by zvecr