qmk/qmk_firmware #10288: [Feature Request] Please add how to configure "U" shaped coulumns in the manual

Supermagnum, 2020-09-10:
There are no examples of how to do "U" shaped coulumns in the manual, it is a great way to conserve used controller pins.
A example is this schema.
https://github.com/Supermagnum/98keys-ergo/blob/master/left/pcb-flat-left-elec-diagram.pdf

I think that it can be used for rows too.

zvecr, 2020-09-11:
The scenario is already supported, and there are plenty of examples within the repo. Look for boards where the rows are doubled, cols are halved, like the `keebio/dilly`. For reference,  <https://github.com/qmk/qmk_firmware/blob/master/keyboards/keebio/dilly/config.h#L15-L16> and <https://github.com/qmk/qmk_firmware/blob/master/keyboards/keebio/dilly/dilly.h#L6-L17>.

As a note, <https://github.com/Supermagnum/98keys-ergo/blob/master/left/pcb-flat-left-elec-diagram.pdf> is not set up correctly as a duplex matrix, sw1 and sw7 would be the same matrix location (Notice in the "Another example" you provide, there are green and blue rows).

Supermagnum, 2020-09-11:
> The scenario is already supported, and there are plenty of examples within the repo. Look for boards where the rows are doubled, cols are halved, like the `keebio/dilly`. For reference, [...]

Thanks, I will fix it as soon as possible.