qmk/qmk_firmware #10314: Bluefruit Feather 32u4 Wrong Keys over Bluetooth

maberalc, 2020-09-14:
I got a problem with Bluefruit: My keyboard is working wired but not over BT (In fact I'm writing this message with it wired). When I connect it to an iPad or another computer with bluetooth, as you can see in the image, it is doing weird keypresses.
[image]
 It worked before on a test keymap with just 9 keys wired and now with the definitive keymap it's when it is sending those keypresses. I got the Bluefruit Feather not the SPI friend, so it is soldered OK. I used an Arduino sample to clear EEPROM (https://www.arduino.cc/en/Tutorial/EEPROMClear)  as #8061 said clearing EEPROM worked for him, and still doesn't work. I tested the Bluetooth now with Adafruit's HID Keyboard Arduino example and it is working properly too. What could be the problem?
Just to make sure that you know which is my configuration, I'm using an MCP23017 as a port expander to read the columns and is working as intended while wired, and my keymap is using Spanish keymap codes (I don't think it can cause it fail but just saying for if this could be the problem).

maberalc, 2020-09-14:
If you want more info or my config files tell me.

maberalc, 2020-09-17:
No one can help? This happened never before?

pwlandoll, 2020-09-18:
I wish I could offer some help, but I can only say that I'm having the same issue - fine on wired, and those same key presses over Bluetooth. I'm using an Adafruit Feather 32u4 Bluefruit LE for a 40% handwired, config files at https://github.com/pwlandoll/void40. 

#5439 might be related to our problem?

maberalc, 2020-09-18:
In fact it is related (or at least I found the error while reading that issue). When it worked on my board, NKRO was not forced. Problem is NKRO. I just tested it and it is working with 6KRO. I don’t know if it will be solved someday but for now I will just use bootmagic when I want to use BT.

maberalc, 2020-09-18:
I think I should close this by myself

closed 2020-09-18 by maberalc

fauxpark, 2020-09-18:
Neither of the QMK-supported Bluetooth modules (Adafruit BLE and RN-42) can do NKRO, this is a limitation of the modules' firmware itself, not QMK.

maberalc, 2020-09-18:
OK, anyway I found the problem. For people's kmowledge, it would be great to write it on QMK Docs' Bluetooth Page, so people don't get mad about his bluetooth not working OK.

pwlandoll, 2020-09-18:
:facepalm: Disabling NKRO worked for me as well. I'll see what I can do about expanding that documentation page with what we've learned. Thanks @maberalc and @fauxpark !