qmk/qmk_firmware #10386: [Bug] Can't compile preonic:dudeofawesome

NeoScaler, 2020-09-21:
Compiling the @dudeofawesome keymap for the Preonic does not work. Compiling other keymaps works. 
## Describe the Bug

Seems like the dudeofawesome.h file is missing, which is imported in config.h.

```
$ qmk compile -km dudeofawesome
INFO Compiling keymap with make preonic/rev3:dudeofawesome

QMK Firmware 0.10.14
Making preonic/rev3 with keymap dudeofawesome

arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: users/dudeofawesome/dudeofawesome.c                                                     In file included from ./keyboards/preonic/keymaps/dudeofawesome/config.h:5,
                 from <command-line>:
users/dudeofawesome/dudeofawesome.h:10: error: "WORKMAN_SOUND" redefined [-Werror]
     #define WORKMAN_SOUND \

In file included from quantum/config_common.h:337,
                 from ./keyboards/preonic/config.h:21,
                 from <command-line>:
quantum/audio/song_list.h:69: note: this is the location of the previous definition
 #define WORKMAN_SOUND E__NOTE(_GS6), E__NOTE(_A6), S__NOTE(_REST), E__NOTE(_GS6), E__NOTE(_A6), S__NOTE(_REST), ED_NOTE(_FS7), S__NOTE(_REST), ED_NOTE(_A7),

cc1.exe: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:386: .build/obj_preonic_rev3_dudeofawesome/dudeofawesome.o] Fehler 1
Make finished with errors
make: *** [Makefile:584: preonic/rev3:dudeofawesome] Fehler 1

```

## System Information

 - Keyboard:  Preonic
   - Revision (if applicable):  Rev3
 - Operating system: Windows 10 MSYS2
 - QMK Firmware version:
0.10.4
 - Any keyboard related software installed? 
No

## Additional Context

fauxpark, 2020-09-21:
No, it's because `WORKMAN_SOUND` is defined in dudeofawesome.h, but it is already defined in song_list.h. A simple
```c
#ifdef WORKMAN_SOUND
#    undef WORKMAN_SOUND
#endif
```
before it should resolve the error.

FilipParyz, 2020-10-13:
I'm working on the fix for you, I'll get back to you when things move on :)

closed 2020-10-14 by fauxpark

FilipParyz, 2020-10-14:
@NeoScaler Done! My fix was merged to QMK repo so if you pull the newest changes you should be able to compile the dudeofawesome keymap with preonic in version 3 :)