qmk/qmk_firmware #10432: [Feature Request] WS2812 simultaneus RGB matrix + underglow

gabbla, 2020-09-24:
To keep PCB manufacturing price low it's a good idea to make them as 2 layers only. Such amount of layers is more than enough if you do not plan to add anything but a matrix plus some other not so invasive feature (underglow, eeprom, piezo speaker). Another cool feature is the RGB matrix. Such feature can be easy to route if we use serial LEDs (like ws2812-2020) and can be fitted in 2 layers, or can be more complex to route if we use individual PWM-controlled LEDs (via dedicated chips). As now, using WS2812 LEDs for both underglow and RGB matrix is not possible since them are mutually exclusive (strange behavior if enabled together).

**TL;DR** Adding the option to chain underglow and RGB matrix using the same (or 2 different) WS2812 setup.

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

It would be cool to enable the PCB designer to use serial RGB LEDs for both underglow and matrix in order to reduce complexity (less chip to solder, less trace to route) and costs (less chip to buy, possibly less layer in the final PCB).

I see 2 possible approaches: 
- Using 2 separate data out PINs
- Using a shared pixel buffer between matrix and underglow

I'd like to discuss and in case contribute to such feature (got hardware to test it).

Gab

elfmimi, 2020-10-11:
QMK have already got enough functionality to support what you are planning to do.

It's just you need to define how you want it to behave. Which means you have to define new set of keycodes and corresponding 'process' routine to control rgb_matrix and underglow separately or whatever.

Refer to [rgbkb/sol/rev2](https://github.com/qmk/qmk_firmware/blob/master/keyboards/rgbkb/sol/rev2/rev2.c) .
This keyboard controls on-and-off state of rgb_matrix and underglow independently.
https://github.com/qmk/qmk_firmware/blob/cacf495c91c7f6ad0e31edcefa8355eba98f75b6/keyboards/rgbkb/sol/keymaps/default/keymap.c#L134-L158

rgbkb/sol/rev2 is using "shared pixel buffer" strategy with WS2812 driver.
https://github.com/qmk/qmk_firmware/blob/cacf495c91c7f6ad0e31edcefa8355eba98f75b6/keyboards/rgbkb/sol/rev2/rules.mk#L19

If you what to do something else, you can override colors of each and every LEDs by writing rgb_matrix_indicators_kb() .
( yeah, it says indicator but it should be okay to do underglow stuff too. )

It is only after you implement something generally useful at keyboard level, that people start to talk about porting it into core feature. I guess that is the idea here.

gabbla, 2020-10-13:
Thank you very much! I'll dig more in the rgb code section, starting from you incipit. I don't know why I completely ignored the LED flags.

gabbla, 2020-10-20:
Ok, what I was trying to achieve was controlling underglow + matrix independently, but to be honest it does not make sense at all after seeing how well the effects blend. I was not aware of this functionality, thanks @elfmimi !

closed 2020-10-20 by gabbla