qmk/qmk_firmware #10435: [Feature Request] LAYOUT_alice addition

stanrc85, 2020-09-24:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

MudkipMao, 2020-10-02:
I already have one made, so I can take this up pretty easily

stanrc85, 2020-10-03:
> I already have one made, so I can take this up pretty easily

I can help as well, I just didn't want to do the work first since I think it would involve updating a lot of peoples keymaps.

MudkipMao, 2020-10-03:
I don't think you're required to updated any other people keymaps, you only need to create the layout and put it in the layouts/default/ folder as an option for people to use in the future

closed 2020-10-27 by stanrc85