qmk/qmk_firmware #10448: Help with rotary encoder (pro micro)

boufe16, 2020-09-26:
Hello, 

I'm writing a keyboard layout for a custom macropad with a rotary encoder.

My issue is that the rotary encoder is only registering every other "notch" in any given direction direction. 

I have the pins a/b connected to pins F5 and F6 on my promicro
`void encoder_update_kb(uint8_t index, bool clockwise) {
    if (index == 0) { /* First encoder */
        if (clockwise) {
            tap_code(KC_LEFT);
        } else {
            tap_code(KC_RIGHT);
        }
    } 
}` -- this is my code to handle the event

I've played around with the resolution value but nothing seems to work... the part itself worked fine in arduino so I think I'm misconfiguring something in the software. 

any ideas? 
Thanks for your time

fauxpark, 2020-09-26:
Sounds like you need to change the resolution.
https://docs.qmk.fm/#/feature_encoders

boufe16, 2020-09-27:
Thanks, i reread the doc I was using the wrong value for resolution indeed.

closed 2020-09-27 by boufe16