qmk/qmk_firmware #10478: Deprioritize beta.docs.qmk.fm and promote docs.qmk.fm

ymotongpoo, 2020-09-29:
Currently, the search results on major search engines show beta.docs.qmk.fm (hereafter, the beta) pages in the first pages and don't docs.qmk.fm (hereafter, the official). This causes additional work on finding the pages in the corresponding pages on the official, say typing "docs.qmk.fm" in the URL bar and finding the corresponding chapter/section by the search feature on the official page.

I do not know the background of both existing, but the official should be more visible than the beta because it has translated versions and it's easy to navigate non-English speaking/reading developers. Could you consider and try promoting the official and deprioritize the beta?

The following scresnshot is the example from Google with query "qmk rgb lighting".

https://yabumi.cc/174d76b82d6efe820d8d71dd.png

ymotongpoo, 2021-01-05:
Can anyone triage this issue?

kb-elmo, 2021-01-19:
The official docs run on Docsify. Not Gitbook. 

Quote from the Discord:
> gitbook generates a static site, docsify does not, so google is only able to index gitbook

tl;dr: Google is not able to index docsify completely and will only ever return the startpage of the docs.

ymotongpoo, 2021-01-20:
@kb-elmo I meant on https://github.com/qmk/qmk_firmware/issues/10478#issuecomment-704289314 was "beta" one is running on Gitbook, which causes the issue.

As far as I searched about Docsify, it seems now indexing is supported.
https://github.com/docsifyjs/docsify/issues/231#issuecomment-748824590

skullydazed, 2021-02-03:
This has been a long-standing issue that we have tried several times to rectify, but so far it's turned out to be more work than expected.

Due to https://github.com/docsifyjs/docsify/issues/238#issuecomment-474814120 we can't use router mode, unless we want to move the docs away from github pages. We would move to a static site generator but that's a fair chunk of work that several people have started but never finished. There are several pieces that need to be in place before we flip the switch:

* Static site generator identified, setup, and configured
* Github actions to actually do the generation
* Replace the docsify-specific things (like tip/alert boxes)
* Theme it so the docs look nice and look like a QMK project
* Make sure `qmk docs` continues to work
* Probably more I'm forgetting

I would say this is at the top of our "things we want haven't had time for" list. Help getting over these humps would be greatly appreciated.

closed 2022-08-29 by zvecr