qmk/qmk_firmware #10540: [Bug] BOOGMAGIC_ENABLE = full not working when VIA is enabled

ifohancroft, 2020-10-04:
# Describe the Bug

When creating a new keymap, if you enable VIA in the keymap's rules.mk and you set BOOTMAGIC_ENABLE to full, BOOTMAGIC_ENABLE works as 'lite'.

## System Information

 - Keyboard:  Idobo (ID75)
 - Operating system: Manjaro Linux 64bit with kernel version 5.8.6
 - AVR GCC version: 10.2.0
 - ARM GCC version: 10.2.0
 - QMK Firmware version: 0.10.22

 - Any keyboard related software installed?  None

Wilba6582, 2020-10-04:
It's not a bug, it's a feature. 😂

fauxpark, 2020-10-04:
Bootmagic Full is not as reliable as Lite, because it requires that the default layer upon startup contains, or allows access to, at least the space and B keys to get into bootloader mode. Lite, on the other hand, does not care which keys are mapped; it goes purely off of the switch matrix positions, and additionally clears the EEPROM. It's also a much simpler feature (= less code size) so I imagine we may at some point phase out Bootmagic Full, as there's not really that much need for all the extra bells and whistles it has.

zvecr, 2021-09-01:
Closing as full was removed in the recent develop cycle <https://docs.qmk.fm/#/ChangeLog/20210828?id=bootmagic-full-removal>.

closed 2021-09-01 by zvecr