qmk/qmk_firmware #10556: [Bug] Playing audio in music mode more than  11000 frequency render my keyboard useless

karnadii, 2020-10-05:
## Describe the Bug

the music mode will make the keyboard laggy or useless when playing high frequency (more than 11000) in music mode. my fix is to map audio changing the last key to the low number so the last key not generating more than 11000 frequency.
```
const uint8_t music_map[MATRIX_ROWS][MATRIX_COLS] = KEYMAP(
       59, 60, 61, 62, 63, 64,     0,  1,  2,  3,  4,  5,  6,  7,
       45, 46, 47, 48, 49, 50,    51, 52, 53, 54, 55, 56, 57, 58,
       31, 32, 33, 34, 35, 36,    37, 38, 39, 40, 41, 42, 43, 44,
   16, 17, 18, 19, 20, 21, 22,    23, 24, 25, 26, 27, 28, 29, 30,
0,  1,  2,  3,  4,  5,  6,  7,     8,  9, 10, 11, 12, 13, 14, 15
);

```
there should be a way to limit the max frequency allowed in music mode. like the clicky mode. 
## System Information

 - Keyboard:  qsplit_mod (https://github.com/karnadii/qmk_firmware/tree/qsplit/keyboards/handwired/qsplit_mod)
   - Revision (if applicable):  
 - Operating system: MacOs
 - AVR GCC version: 8.4.0

 - ARM GCC version: 8.3.1

 - QMK Firmware version: 0.10.25

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

FilipParyz, 2020-10-13:
11kHz is a lot for a keyboard that has to do other task as well as playing music.
From what I understand you want a way to limit the maximum frequency (to 8kHz let's say), so your board wouldn't go higher and brick itself, right?

karnadii, 2020-10-13:
Yeah, qmk automatically generate the frequency according to col and row number. There should be a way to limit the maximum frequency qmk can play on music mode. So a keyboard with a lot of key like mine will keep working while using music mode without mapping the sound. Or maybe put a notice in the doc how much freq you can play in music mode and some work around if the same thing happen to other user.

A newbie like me will spend days wondering why music mode keep breaking the keyboard. It turn out the frequency qmk played ia too high.