qmk/qmk_firmware #10587: [Bug] TAPPING_TERM_PER_KEY breaks PERMISSIVE_HOLD with LT()

sambaum, 2020-10-09:
## Describe the Bug
When using LT() with PERMISSIVE_HOLD enabled, a key combo works as expected: The hold action registers when the tapping key is released (within tapping term).

When enabling TAPPING_TERM_PER_KEY, the behavior changes. It is enough to just define TAPPING_TERM_PER_KEY in config.h (no usage of the get_tapping_term() function is required)
Now the hold action only registers if the tapping term expires, as if PERMISSIVE_HOLD is not enabled.

## System Information

 - Keyboard:  mb60rgb_iso
   - Revision (if applicable):  
 - Operating system: Windows
 - AVR GCC version: 5.4.0

 - ARM GCC version: arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 - QMK Firmware version: 0.10.9

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

Tested with 499 tapping term and a default keymap without any extras.

I didn't test if other dual-role keys are affected.

To me TAPPING_TERM_PER_KEY gains in importance, because ACTION_TAP_DANCE_FN_ADVANCED_TIME is deprecated in favor of TAPPING_TERM_PER_KEY

sambaum, 2020-11-03:
I had a look at the code and realized that if I set TAPPING_TERM globally or TAPPING_TERM_PER_KEY for the relevant key to 500, the problem is not there anymore. So is this intentional? I don't see a problem setting it to 500, it's just not documented. I couldn't find any info about why this threshold of 500ms exists. I don't know if this is a bug, or just something that should be documented.

wolfish17, 2020-11-17:
I think a workaround is to define PERMISSIVE_HOLD_PER_KEY and have the function get_permissive_hold(uint16_t keycode, keyrecord_t *record) always return 'true'.

sambaum, 2020-11-20:
Thanks for the tip, but unfortunately it didn't work. What i did now, is removing the following lines from action_tapping.c
```
#        ifdef TAPPING_TERM_PER_KEY
                    (get_tapping_term(get_event_keycode(tapping_key.event, false), keyp) >= 500) &&
#        endif
```
Now everything works fine and it's not causing any issues with my configuration. I have no idea why this check was there in the first place and I was not able to find the reason behind this 500ms condition.

drashna, 2020-12-01:
It's there, because for any tapping term over 500, permissive hold is forcibly enabled.  If you check the defines, the same thing happens without the per key version.  So this check was added to ensure that behavior remains.  Removing it ... may be problematic.

wolfish17, 2020-12-01:
That is very helpful information!   For debugging I was setting tapping term to 1000 to test if PERMISSIVE_HOLD was functioning as expected.  No wonder I was often confused by the behavior I observed.

sambaum, 2020-12-01:
I understand that removing it, may be problematic in some cases. I tried to figure out why this was introduced in the first place and I can see that this condition goes way way back (couldn't figure out the reason though). Either way, forcefully enable permissive hold at 500 isn't a problem in general. I think the conditions in that code section don't really support all combinations of configurations. And why would the behavior change, just by defining TAPPING_TERM_PER_KEY? I can't imagine this is intentional, because the actual tapping term is not changed.

There is probably a way to keep the forceful permissive hold, but without breaking it when using TAPPING_TERM_PER_KEY. But why keep something, if no one knows the reason for it's existence? Reducing complexity would be beneficial too. If someone looses the permissive hold behavior due to this change, they can fix it by defining PERMISSIVE_HOLD. Explicit enabling/disabling seems better to me, rather than some hard-coded threshold.

cmvkk, 2021-02-06:
I just ran into this bug today as well and wanted to recommend the following replacement code:
```
#    if defined(TAPPING_TERM_PER_KEY) || (TAPPING_TERM >= 500) || defined(PERMISSIVE_HOLD) || defined(PERMISSIVE_HOLD_PER_KEY)
                else if ((
#        if defined(PERMISSIVE_HOLD_PER_KEY)
                             get_permissive_hold(get_event_keycode(tapping_key.event, false), keyp)
#        elif defined(PERMISSIVE_HOLD)
                             1
#        else
                             0
#        endif
                             ||
#        if defined(TAPPING_TERM_PER_KEY)
                             (get_tapping_term(get_event_keycode(tapping_key.event, false), keyp) >= 500)
#        elif (TAPPING_TERM >= 500)
                             1
#        else
                             0
#        endif
                                 ) &&
                         IS_RELEASED(event) && waiting_buffer_typed(event)) {
                    debug("Tapping: End. No tap. Interfered by typing key\n");
                    process_record(&tapping_key);
                    tapping_key = (keyrecord_t){};
                    debug_tapping_key();
                    // enqueue
                    return false;
                }
#    endif
```
Points:
- Puts PERMISSIVE_HOLD and TAPPING_TERM checks into an OR block, allowing the behavior to occur if either is true.
- PERMISSIVE_HOLD_PER_KEY overrides PERMISSIVE_HOLD and TAPPING_TERM_PER_KEY overrides TAPPING_TERM.  It's not clear to me that this is the intended behavior just from the docs but it seems intuitively correct to me.
- Not related to this bug, but the "not" (!) is removed from get_permissive_hold per bug #8999.

Would this fix constitute a breaking change?

purple-rw, 2021-03-05:
I just started exploring home row mods (and compiling qmk), and got frustrated with permissive hold not working.  Found the issue, and my fix is functionally the same as shown by @cmvkk.

sambaum, 2021-03-27:
#12125 fixed it.

closed 2021-03-27 by sambaum