qmk/qmk_firmware #10602: [Bug] qmk_firmware is too old or not set up correctly

quantum-booty, 2020-10-10:
I cloned the repository using `git clone git@github.com:qmk/qmk_firmware.git`
But when I do `qmk setup`, it returns the following error:
```
Error: /home/henryw/qmk_firmware is too old or not set up correctly!
Please update it or remove it completely before continuing.
```

Shadowtrance, 2020-10-10:
Yep I'm getting this too.
Just updated my local (already set up) copy because i was a fair bit behind and now I'm just getting this when I try compile or setup or anything really. :(

fauxpark, 2020-10-10:
This is probably due to https://github.com/qmk/qmk_firmware/pull/8817.

Run `python3 -m pip install pygments` and you should be good.

Shadowtrance, 2020-10-10:
That did the trick!

closed 2020-10-10 by quantum-booty

TheWeirdDodo, 2020-10-20:
> This is probably due to #8817.
> 
> Run `python3 -m pip install pygments` and you should be good.

Tried this, didn't work for me

chris-ryu, 2020-11-27:
Even after "brew reinstall qmk" the problem is not gone.

gwillad, 2021-01-06:
> 
> 
> > This is probably due to #8817.
> > Run `python3 -m pip install pygments` and you should be good.
> 
> Tried this, didn't work for me

didn't work for me either

fauxpark, 2021-01-06:
@TheWeirdDodo @chris-ryu @gwillad I've updated my comment, try it now.

gwillad, 2021-01-06:
I used `util/install_qmk.sh` and that fixed it

chris-ryu, 2021-01-06:
In my case, brew android-sdk location was the problem.
sudo ln -s /usr/local/Caskroom/android-sdk /usr/local/share/android-sdk
after this linking, the problem was resolved

fauxpark, 2021-01-06:
The android SDK has nothing to do with QMK.