qmk/qmk_firmware #10651: [Bug] QMK_HOME path setup incorrectly, cannot fix

ChrisIck, 2020-10-15:
Apologies if this is a noob question. I am setting up QMK on a linux subsystem of a windows device. In setting up the QMK environment using `qmk setup`, I accidentally set the home directory one level up (`/mnt/c/Users/Username/`, which contains `qmk_firmware`).

It should be a simple case of updating QMK_HOME manually, but setting it manually (`export QMK_HOME="/mnt/c/Users/Username/qmk_firmware"`) doesn't resolve the issue. Attempting to change it via `qmk config`, or calling the qmk command at all yields the following error:

```
Warning: /mnt/c/Users/Username does not end in "qmk_firmware". Do you need to set QMK_HOME to "/mnt/c/Users/Username/qmk_firmware"?
Error: /mnt/c/Users/Username is too old or not set up correctly!
Please update it or remove it completely before continuing.
```
How can I manually change the `qmk_home` config to avoid this message?

fauxpark, 2020-10-15:
Coukd you run `python3 -m pip install -U qmk` and then try again? A recent PR added a new dependency to the CLI on the qmk_firmware side, so the Python package needs to be updated.

ChrisIck, 2020-10-15:
I updated it last night, and tried again just now to no avail; I'm already on the latest release. Thanks for your suggestion though.

fauxpark, 2020-10-15:
Okay, how about `python3 -m pip install pygments`?

ChrisIck, 2020-10-15:
I tried this per issue #10602, also on the latest release of `pygments`. Just updated `pip` and tried updating both `qmk` and `pygments`, no changes, qmk still broken.

Is there a default location to an `.ini` file that contains the `QMK_HOME` configuration that I can manually edit?

fauxpark, 2020-10-15:
Yep, I believe it's in `~/.config` somewhere.

ChrisIck, 2020-10-15:
Manually edited `~/.config/qmk/qmk.ini`, am able to sucessfully setup/configure qmk now, thanks for your help!

closed 2020-10-15 by ChrisIck