qmk/qmk_firmware #10675: [Feature Request] Can I change the rgb_matrix_drivers.c file and upload it?

XBowsTech, 2020-10-17:
## Feature Request Type

- [ ] Core functionality
- [·] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [·] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description
I checked the code in the RGB_MATRIx_drivs.c file and found that it only supports two IS31FL3731 chips working at the same time.
Because my new keyboard needs to use four IS31FL3731 chips at the same time, So I want to modify and upload the RGB_MATRIx_drivs.c file so that it can support four IS31FL3731 chips to work at the same time.
So far I have completed the code modification.Through the test, its working condition is good.So I want to upload my modified file. What should I do?
Also, I'm working on how to add my keyboard to QMK.
（I'm a green hand.）

FilipParyz, 2020-10-17:
You should create a pull request, after the review from core contributors/maintainers, you'll either be requested to change some things or your changes will be merged 👍

elfmimi, 2020-10-18:
it's dup.
https://github.com/qmk/qmk_firmware/issues/6905

closed 2021-02-15 by stale[bot]